OpenBCI/OpenBCI_GUI #403: error in Linux script lline 5

cosmobille, 2018-11-14:
## Problem error in bash script

Explain the problem  Error in line 5 ` $APPDIR/java/bin/java`  Script won't run as is. 
Should be `$APPDIR/java/java/bin/java`
## Expected  fix script

In a perfect world, what do you expect to happen. fix script

## Operating System and Version Linux Mint 19

Linux

## GUI Version 3.3.2

The version is displayed on startup.

## Running standalone app

Are you running the downloaded app or are you running from Processing 3

## Type of OpenBCI Board

Cyton/Cyton+Daisy/Ganglion

## Are you using a WiFi Shield?

Yes/No

cosmobille, 2018-11-14:
Error in line 5 $APPDIR/java/bin/java Script won't run as is.
Should be $APPDIR/java/java/bin/java

retiutut, 2018-11-15:
This is the same problem as #371

andrewjaykeller, 2018-11-15:
@cosmobille did v4.0.2 solve this problem? https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v4.0.2

cosmobille, 2018-11-15:
oh ya. I just wanted to make people aware of it.
thanks, Jeffrey

daniellasry, 2019-04-19:
Fixed in current version of the GUI

closed 2019-04-19 by daniellasry