OpenBCI/OpenBCI_GUI #405: Cannot connect to ganglion over bluetooth on OSX Mojave

rschaef1, 2018-11-15:
## Problem

Explain the problem

## Expected 

In a perfect world, what do you expect to happen.

## Operating System and Version

macOS/Windows/Linux

## GUI Version

The version is displayed on startup.

## Running standalone app

Are you running the downloaded app or are you running from Processing 3

## Type of OpenBCI Board

Cyton/Cyton+Daisy/Ganglion

## Are you using a WiFi Shield?

Yes/No

rschaef1, 2018-11-15:
I have been using my Ganglion board for some time now, and it has always worked perfectly. Unfortunately, when I updated my MacBook to Mojave yesterday, the board just stopped connecting. I have tried GUI 3.0, 3.2, and 2.2, and the Ganglion board doesn't show up as a connection option on any of these. Any Ideas?

andrewjaykeller, 2018-11-15:
Use: 
https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v4.0.3-beta.0

closed 2018-11-15 by andrewjaykeller