OpenBCI/OpenBCI_GUI #406: OpenBCI GUI Runtime Exception

HamzaHummam, 2018-11-16:
## Problem

Whenever I try to execute the OpenBCI GUI application or Code its not responding, and following error is displayed in the log.
java.lang.RuntimeException: Waited 5000ms for: <4dac438e, 436bb44b>[count 2, qsz 1, owner <main-FPSAWTAnimator#00-Timer0>] - <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1>
	at processing.opengl.PSurfaceJOGL$2.run(PSurfaceJOGL.java:410)
	at java.lang.Thread.run(Thread.java:748)

## Expected 

Successful Execution of OpenBCI GUI 

## Operating System and Version

Windows 10  and Mac OS

## GUI Version

4.0.2

## Running standalone app

I have tested both standalone app and also tried the code both are not responding

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

Want to use wifi Shield.

andrewjaykeller, 2018-11-16:
Thanks for the bug report! Trying to duplicate, what version of Mac OS?

HamzaHummam, 2018-11-23:
Mac OS 10

rivasd, 2018-12-07:
I have the same issue, version 4.0.3, running in Processing 3.4 on Win10

andrewjaykeller, 2018-12-07:
Hey! I was able to duplicate this bug when my computer was way over powered and not able to allocate enough resources for the GUI. Are you running other programs in the background that are hogging the CPU?

commit 6094a3c5 (2019-02-09): Merge pull request #433 from daniellasry/fix-timeout

[FIX] the GUI timing out on launch. fix #409 fix #406 fix #426
commit 661d74ee (2019-02-09): [FIX] the GUI timing out on launch. fix #409 fix #406 fix #426
closed 2019-03-09 by daniellasry