OpenBCI/OpenBCI_GUI #407: Indication of SW Version

jps2000, 2018-11-20:
## Problem
Software version number is  shown during startup but only very short time
Seem to depend on start up speed / Windows version ( time is not constant???)
pls check

In a perfect world, what do you expect to happen.

display it for 0.5 - 1  second at least or show it permanently somewhere

Windows

Windows 10 seems worse than W7  or is it dependent on PC speed ???
## GUI Version
4.0.3

## Running standalone app

downloaded app 
## Type of OpenBCI Board

n/a
## Are you using a WiFi Shield?

No

retiutut, 2018-11-20:
I could easily add the version to the hover text of the "Update" button.

andrewjaykeller, 2018-11-20:
Good idea!

retiutut, 2018-11-20:
[image]

andrewjaykeller, 2018-11-20:
!

andrewjaykeller, 2018-11-20:
PR?

jps2000, 2019-02-11:
solved in 4.0.1

closed 2019-02-11 by jps2000