OpenBCI/OpenBCI_GUI #409: OpenBCI GUI will not run

stemcubator, 2018-11-21:
## Problem

OpenBCI GUI v4.0.3-beta.0 will not launch in Windows 10

## Expected 

Run the program

## Operating System and Version

Windows 10 64 bit 

## GUI Version

OpenBCI GUI v4.0.3-beta.0

## Running standalone app

Yes I am run the stand alon app

## Type of OpenBCI Board

Ganglion

## Are you using a WiFi Shield?

No

jps2000, 2018-11-21:
Had same problem on my W10 installation ( use also W7 elsewhere)  but was not sure what it caused. ( hang with gray screen during start)
Throwed then completely away old version including Hub and re-installed everything including zadig driver installation and shortcuts.
This made it work.

jps2000, 2018-11-26:
UNfortunately now it does not work any more  so there is an issue.
Starts with light grey than turn a little bit darker and it hangs.
Can you pls list in the dark screen what the program is doing during loading to trace back where it hangs?
Is there another way to debug?

commit 6094a3c5 (2019-02-09): Merge pull request #433 from daniellasry/fix-timeout

[FIX] the GUI timing out on launch. fix #409 fix #406 fix #426
commit 661d74ee (2019-02-09): [FIX] the GUI timing out on launch. fix #409 fix #406 fix #426
closed 2019-03-09 by daniellasry