OpenBCI/OpenBCI_GUI #410: Accelerometer widget

jps2000, 2018-11-21:
## Problem

Accelerometer window in streaming is  15 sek;  in playback it is 3 sek. It can not be changed

## Expected 

# Window  should go together with time series --> add adjustment in acc widget and synchronize selection with time series widget.
# Window should have same dimension and location as time series (to appear aligned when  time series and acc are arranged vertically. Hence also vertical scale  of acc may be left instead of right.
Remark: with >4 channels vertical arrangement may not be possible but it will look better anyhow I think
# Add horizontal scale (like time series)

Remark: Time series window should have 10 sec instead of 7 sec  as largest window ( steps should double previous value (1,2(3),5,10, 20....) 

## Operating System and Version

W7

## GUI Version

4.0.3

## Running standalone app

Are you running the downloaded app or are you running from Processing 3

## Type of OpenBCI Board

N/A

## Are you using a WiFi Shield?

N/A

retiutut, 2018-11-21:
> Window should have same dimension and location as time series (to appear aligned when time series and acc are arranged vertically. Hence also vertical scale of acc may be left instead of right.

Like this?
[image]

> Remark: Time series window should have 10 sec instead of 7 sec as largest window ( steps should double previous value (1,2(3),5,10, 20....)

Great suggestion! A 10 second option can definitely be added. This paper suggests the same type of epoch lengths: https://www.ncbi.nlm.nih.gov/pubmed/27137952

jps2000, 2018-11-21:
Same width is just nice to have.  i mentioned it just for completeness (I consider Acc also as a kind of time series)

retiutut, 2018-11-24:
@jps2000 Can you please change the title to something like "Sync horizontal scale of Time Series and Accelerometer widgets" ?

retiutut, 2018-11-24:
[image]

retiutut, 2018-11-24:
[image]
Also, #247 has just been fixed!!! 💯

retiutut, 2018-11-25:
[image]

jps2000, 2018-11-26:
Great.  Thank you very much.
Is it so that window is adjusted in time series and affects both widgets? ( in stream and playback)
I do not know how much time you can / like to invest more
Would be nice to have same light and dark grey bar with at least vert scale adjustment . 
Vert scale +/-( 1),2,4. ( mostly it will be in the range +/-1.

retiutut, 2018-11-26:
> Great. Thank you very much.
> Is it so that window is adjusted in time series and affects both widgets? ( in stream and playback)
> I do not know how much time you can / like to invest more
> Would be [...]

Yes, the functions are there but not being used yet. I need to get the data displayed on the graph first.

andrewjaykeller, 2018-11-26:
@retiutut look at the analog widget for inspiration!

retiutut, 2018-11-26:
@andrewjaykeller mind reader!

retiutut, 2018-11-26:
[image]

jps2000, 2018-11-27:
You are  progressing. Great job.
This demo triggers another suggestion. 
Clear graphs in time series and acc at starting point of the  playback bar
Alternatively put a marker (vertical line or bar) there
Another great thing would be a horizontal (time) scale under the playback graph or just indicating  begin and end time.
Can open a new issue if you support this ideas.

jps2000, 2018-11-27:
> 
> 
> [image]

jps2000, 2018-11-27:
How can I access the analog widget?

retiutut, 2018-11-27:
> How can I access the analog widget?

With a Cyton board in Live mode

> Clear graphs in time series and acc at starting point of the playback bar
> Alternatively put a marker (vertical line or bar) there

This may need to be done:
- [ ] Clear graphs in time series and acc at starting point of the playback bar

> Another great thing would be a horizontal (time) scale under the playback graph or just indicating begin and end time.
Look at the status bar of the bottom of the GUI while in Playback Mode!

retiutut, 2018-12-01:
> Window should have same dimension and location as time series (to appear aligned when time series and acc are arranged vertically. Hence also vertical scale of acc may be left instead of right.

- [x] Match position and width of Time Series and Accelerometer Gplots

> Window should go together with time series --> add adjustment in acc widget and synchronize selection with time series widget.

- [ ] Add Accelerometer dropdown menus to SoftwareSettings
- [ ] Make a "Sync Duration" button in Time Series to allow the option to change the duration of Time Series, Accelerometer, and Analog Read simultaneously

retiutut, 2018-12-01:
[image]

jps2000, 2018-12-02:
You are doing great.  It was significant effort bit it improved tremendously.
Just had another idea.  you may consider to have acc x on, acc y on, acc z on separately. sometimes just one axis is interesting. 
But that's just in case you face boring times....
Decision to do something is  interest * priority

jps2000, 2018-12-03:
Great progress.
However it would be more logical to have the sync button in the acc widget.
But you invested already plenty of time in that stuff.
As I do not see applications not to sync windows.
The playback bar is appearing in time series widget only ( rightfully). Hence considering time series as master,it would be most easiest to skip the sync button ( leaving sync = on behind) and skip the window button in acc widget. 
Then we are there with littlke further headache.
I suggest to stop than this topic and invest time better  in other things.

(I have plenty of other ideas (un)fortunately.)
But anyhow applause applause and thank you for your effort.

retiutut, 2018-12-03:
@jps2000 No worries, you have such great and workable suggestions!!! 🥇 

>  However it would be more logical to have the sync button in the acc widget.
> But you invested already plenty of time in that stuff.

I put it in Time Series because you can't view Accelerometer and Analog Read data at the same time, I think. So I would have had to put a "Sync" dropdown in both Accel. and Analog Read widgets.

> As I do not see applications not to sync windows.
Then I will set the default option to "On"!

> Hence considering time series as master,it would be most easiest to skip the sync button ( leaving sync = on behind) and skip the window button in acc widget.

I think someone might want to have the option to turn it off. But I can make it so that "Sync" is "On" 100% of the time, unless users changes/saves the setting.

> Pls have a look at the the signals they seem to be offset about 1/2 second??
Yes, this is a problem. https://github.com/jagracar/grafica/issues/11

retiutut, 2018-12-03:
Another solution would be to add a "sync" option to the Window dropdown of both Accelerometer and Analog Read.

retiutut, 2018-12-03:
[image]

jagracar, 2018-12-03:
Hi,

Are you using this method to set the frame rate?
https://processing.org/reference/frameRate_.html

I still think is not a problem of the grafica lib.

jps2000, 2018-12-04:
To be more specific:
when you change the framerate then the x axis difference wetween time series  and acc changes.
But this difference remains constant.

BTW I do not know what this change in framerate does neither what it it is good for nor how do I change it

retiutut, 2018-12-04:
I've been looking at this problem for days. It does what it does and it isn't right.

jps2000, 2018-12-04:
Still not clear what you are saying.    lets assume time series are displayed real time. There are ticks in 1 second distance. If you make also ticks  with acceleration  accelerometer is displayed  not synchronously?
the 1 sec ticks appears stretched or compressed or just shifted? Ant the shift is constant or  increasing?

retiutut, 2018-12-04:
[image]

jps2000, 2018-12-05:
Thanks
I will test it once  your synchronized time frame  software is released.
Just to clarify: The frame rate determines the refreshing of the GUI widget?
Why do I need to change that?
There is likely another acc issue. Maybe only in ganglion. It seems that acc enters sleep mode when not moved fast enough. But I will come back on that later. I think this is a peripheral software issue and not one of the GUI

retiutut, 2018-12-18:
Tested using X axis only.

Thanks for the guidance @andrewjaykeller! 🔥

[image]

andrewjaykeller, 2018-12-18:
Looks good @retiutut that's amazing you just did that.

jps2000, 2018-12-18:
yeah
Huge huge improvement!!!!!!!!
Have you tried to hold ganglion constant e.g. on the table for some seconds and then turn slowly to the side?
Is the acc following also small turns or do you need to shake it first to get an reaction?

retiutut, 2018-12-18:
lol no shaking necessary I promise
[image]

jps2000, 2018-12-19:
Not sure but is there no zero line?
like to see zero marked at least . better add a grid /horizontal lines please

retiutut, 2018-12-19:
[image]

retiutut, 2018-12-19:
Done. As simple as changing:

```
plot.drawGridLines(0);
```
to
```
plot.drawGridLines(2);
```

[image]

jps2000, 2019-02-11:
Solved in 4.0.1 
Thank you looks great now

closed 2019-02-11 by jps2000