OpenBCI/OpenBCI_GUI #412: OpenBCI v4.0.2 Not Work in Debian GNU/Linux.

wellton, 2018-11-24:
## Problem

The OpenBCI GUI v4.0.2 not Work in Debian GNU/Linux..  (source)

## Expected 

I open the Processing 3.4 and open the OpenBCI GUI v4.0.2 and the error below appear. =(

## Operating System and Version

Debian GNU/Linux

[image]
Linux

## GUI Version

v4.0.2

## Running standalone app

Are you running the downloaded app or are you running from Processing 3.4

## Type of OpenBCI Board

Cyton+Daisy

## Are you using a WiFi Shield?

Yes

andrewjaykeller, 2018-11-24:
Did you run the hub?

wellton, 2018-11-25:
Hello Andrew. 

Yes, I run the hub...  see.

[image]

winsteadworks, 2018-11-26:
@wellton try downloading from here: https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/3.4.0

BTW, to get it to work for Ubuntu 18.04. You need to run these commands
sudo apt install libgconf2.4

then go to your application.linux64 folder 
Make sure to run the GUI as sudo ./OpenbciGUI. Without sudo the software won't run correctly

winsteadworks, 2018-11-26:
@wellton try downloading from here: https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/3.4.0

BTW, to get it to work for Ubuntu 18.04. You need to run these commands
sudo apt install libgconf2.4

then go to your application.linux64 folder 
Make sure to run the GUI as sudo ./OpenbciGUI. Without sudo the software won't run correctly.

Also this may be obvious to some, but make sure you download java!

sudo apt install default-jre

wellton, 2018-11-27:
@winsteadworks  thanks =) but, I want to use the v4.0.2 =/

daniellasry, 2019-01-05:
Hey @wellton , are you trying to run the OpenBCI GUI on a raspberry pi? The errors you are getting seems consistent with running on a pi.

I'm afraid the GUI cannot run on raspberry pi because it uses some graphics drawing features than are not supported on the device. Others have attempted it, but without success. See issue #146 for example. It's possible you could get the GUI partially running if you modify the code to bypass these graphics, but there's no guarantee you won't run into something else or beak the app.

wellton, 2019-02-06:
The @daniellasry fixed this problem =) 
THAAAANKKSSSSSS

https://github.com/daniellasry/OpenBCI_GUI/commit/661d74ee3a6bc3db2787b88451683567b6e2f3b5

closed 2019-02-06 by wellton