OpenBCI/OpenBCI_GUI #416: Enhancement: New widget to display L and R power spectrum

jps2000, 2018-12-19:
## Problem

It would be great to have a widget that displays the power spectrum of left and right brain halves.
The spectra should be displayed such that easy comparison of L/R is possible 
--> Display R and L on top of each other but inverted
The Masimo Sedline device has this
http://www.masimo.com/products/continuous/root/root-sedline/ 

https://www.medgadget.com/2018/02/masimo-sedline-for-improved-brain-monitoring-under-anesthesia-fda-cleared.html

## Expected 

Make a similar widget that displays last 30 minutes of the power spectrum like the Masimo does.
There could be also a display highlighting easily difference
Electrodes to be used for L and R should be selectable.
That is an interesting feature for anaesthesia pain assessment+ treatment, drugs/ medication abuse awareness etc.etc.

## Operating System and Version

N/A

## GUI Version

N/A
## Running standalone app

N/A

## Type of OpenBCI Board

N/A

## Are you using a WiFi Shield?

N/A

retiutut, 2018-12-19:
Very cool feature request! I have some experience with spectral processing in Max/PureData, so I will look into this.

retiutut, 2018-12-19:
So we already have FFT processing through the minim library. The Sedline device seems to have a proprietary display of FFT data.

jps2000, 2018-12-19:
Its just a dual spectrogram known from Audio Analysis. They plot also a line showing a Kind of center frequency. Von meinem Huawei-Mobiltelefon gesendet-------- Originalnachricht --------Betreff: Re: [OpenBCI/OpenBCI_GUI] Enhancement: New widget to display L and R power spectrum (#416)Von: Richard Waltman An: OpenBCI/OpenBCI_GUI Cc: Peter ,Author So we already have FFT processing through the minim library. The Sedline device seems to have a proprietary display of FFT data.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/OpenBCI/OpenBCI_GUI/issues/416#issuecomment-448686267",
"url": "https://github.com/OpenBCI/OpenBCI_GUI/issues/416#issuecomment-448686267",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

retiutut, 2018-12-19:
[image]

I ran a song I wrote through this sketch:

https://www.openprocessing.org/sketch/4735/#

jps2000, 2018-12-19:
Sounds good😁, just needs log color scale i thinkVon meinem Huawei-Mobiltelefon gesendet

jps2000, 2018-12-21:
Continued..
This long period information or trend displays may show also EMG ( Bandpass around 30Hz + rectifying) EOG ( Electrooculogram) Bandpass ( ~1Hz + rectifying). there can be also others like traces for breathing pattern etc ( Aux channels)
So what we  have then is a kind of polysomnograph ( see wiki) 
The EOG  events coming from REM  for example are rather short events.
To analyze  these in detail a kind of magnifying display would be great. so  moving over the trend display expands the timespan to about 5 seconds.
Not easy to code however......

jps2000, 2019-08-21:
@retiutut 
would be great to see some progress here.
I believe once this widget is available many will love it as real time trend monitor. Something like this is missing now in OpenBCI.

retiutut, 2019-09-22:
@jps2000 Early prototype work:

[image]

Got it to fit and resize within a widget. Just like before, still using FFT from an audio file to make coding the look of the Spectrogram more fun. Though to note, now using 2 FFTs, one for left channel and one for right channel. We should be able to do the same with EEG signals!

jps2000, 2019-09-23:
To start off  you could  first dedicate channel 1 to L and channel 2 to R.  Frequency scale 0-40Hz linear, L/R horizontally mirrored, Time scale 30 min length (last 30 min), showing actual time ,  Record  button for screen shots.
(jpg or tiff).
In case you want / need to do it different to Masimo you can plot  it vertically ( vertical time axis ,  widget format portrait, Then the power spectrum is growing from the bottom like a hourglass.
This may look quite nice and it it obvious what is L and what R.
What do you think?

retiutut, 2019-09-28:
- [ ] To start off you could first dedicate channel 1 to L and channel 2 to R. 

- [ ] Frequency scale 0-40Hz linear,

- [ ]  L/R horizontally mirrored (In case you want / need to do it different to Masimo you can plot it vertically ( vertical time axis , widget format portrait, Then the power spectrum is growing from the bottom like a hourglass.)

- [ ] Time scale 30 min length (last 30 min), 

- [ ] showing actual time ,

- [ ]  Record button for screen shots. (jpg or tiff).

jps2000, 2019-09-28:
Great
I see no need for log/ lin dropdown and frequency limit change at the moment--> skip
Channel select is more important. For example to have check boxes for channels ( odd ch numbers left, even numbers right. 1,3 left, 2,4 right. You need to sum the selected channels like you do in the FFT widget.  Not more, otherwise you need to do it differently for  ganglion and cyton.

retiutut, 2019-10-22:
Upgrading this to high priority! Planning to work on this soon so the first version of Spectrogram widget can be included with 4.1.7! **I will continue to use this thread as a guide.** 

@jps2000 Thank you for your help thus far!

retiutut, 2019-11-11:
@jps2000 How many points per second do we need? An average FFT of each second?

jps2000, 2019-11-11:
Yes 1 second update is also what Masimo is doing

retiutut, 2019-11-13:
Notes for me for later:

> Yes 1 second update is also what Masimo is doing

@daniellasry helped decouple the scrolling from the framerate 💯 , might even be better as a Java TimerTask (maybe).

> 60*30 = 1800 points/pixels-wide for 30 mins of data

This will delay the release of this widget to GUI 5.0, as we will need some type of simple Image Scaling algorithm. Very doable, but there are already discrepancies when the GUI is resized or you move the spectrogram widget to another container of a different size.

retiutut, 2019-11-30:
Image scaling is done!

retiutut, 2019-11-30:
[image]

retiutut, 2019-12-01:
[image]

jps2000, 2019-12-01:
Hi @retiutut Congratulations for the huge progress.
Let me comment something - of course you may have noticed already some pending issues.

Frequency reading  is probably not correct . You can easily check that by looking for a sharp line at 60Hz.( Mains interference, filter off.)
Then max frequency is too high. Should be 0-40 Hz. To prevent aliasing effects also use low pass filtering.
Max frequency select is probably not needed, The brain waves are always the same. 
What may make sense is a kinda setting of the amplitude to color  relation but this should be logarithmic as in the FFT
Try to make 2  separate boxes for L and R  to have them clearly separated

Masimo indicate  signal problems  by a white bar. This could be done when the amplitudes is out of range. But lets discuss this later- step by step.

Is it "easily" possible to turn the graph vertically (by 90 degree)
Then it would be a kinda waterfall, it is clear what is left and what is right, It differs from Masimo and  for the observer it is more easy to see differences between left and right ( in my opinion, may be others see this differently)

retiutut, 2019-12-01:
I will make the standard Max Freq 40 Hz, options for 250 will be available for testing and other unforeseen applications.

Rather than change how I am drawing pixels, I am simply going to do a `rotate()` on the image, and of course the axes will need to change. 😄

I will keep posting pictures as I continue to meet the requirements listed here.

jps2000, 2019-12-01:
Agree
Make a thick line @ 0 Hz
250Hz is too much ( Sample rate Nyquist  Shannon Theorem) but 100Hz may be ok. Then you can easily check 50/ 60Hz signals coming :-)
You do not need a log scale in frequency
The transfer amplitude to color should be logarithmic thats all
Lets keep posting and commenting

wjcroft, 2019-12-01:
Peter, hi.

I think it's great that Richard is adding the spectrogram capability to the GUI. Long long ago Chip Audette had a hacked version of the GUI that did this. And this new capability has many more features / advantages.

At the same time, if you are building some type of app that you will use in your clinical work, are you aware of some of the VPL Visual Programming Language development environments that allow complete customization of graphics output, recording, labeling, event recognition, advanced DSP operations, etc.? 

The one that comes to mind for ease of use is National Instruments LabView IDE. Click the categories here to view the various DSP and graphing capabilities.

https://zone.ni.com/reference/en-XX/help/371361R-01/lvconcepts/labview_documentation_resources/

[I hope that is displaying in English vs German, somehow NI's stupid IP address tracking is forcing me into German pages.]

National Instruments LabView toolbox has extensive VPL Visual Programming Language capability for all kinds of DSP operations, graphing, etc. etc. And is already Cyton compatible. I know of one neurofeedback company, Zengar / NeurOptimal that has built a very sophisticated app using LabView as a base. This type of customization via VPL widgets and property panels, makes it easy.

For those with academic logins, LabView is essentially free.

https://openbci.com/forum/index.php?p=/discussion/274/labview-openbci-toolkit

Other VPLs are also available that allow easy customization via property boxes that control the operation of element blocks. These are all arranged graphically on the screen with lines indicating data stream flow. VPL's I'm aware of include LabView and: BioEra, OpenViBE, neuromore, NeuroPype, BrainBay, etc.

Regards,

William

retiutut, 2020-01-04:
[image]

Still working on the graph axes, but at least all of the data is being drawn to screen and scaled.

retiutut, 2020-01-05:
[image]

Going to draw a color scale to the right for reference. This is using log base 10 on the left, and linear on the right.

retiutut, 2020-01-06:
[image]

Still need to work on the horizontal axis, but log color scale is integrated with settings!

retiutut, 2020-01-06:
[image]

retiutut, 2020-01-06:
Using sample meditation data file and playback mode with timestamps.
[image]

retiutut, 2020-01-06:
Spectrogram widget is here! There may be bugs that haven't been uncovered, so I'll be releasing this under GUI v5.0.0-alpha. Release coming now!

jps2000, 2020-01-06:
Great! I know it was a huge effort that you had to manage. You did it!
Will try new widget  soon with function generator to see if it works as expected
is it displaying now  channel 1 upper and  and channel 2 lower?

retiutut, 2020-01-06:
>is it displaying now channel 1 upper and and channel 2 lower?
It's fully customizable and the GUI should remember all settings for the widget.

retiutut, 2020-01-06:
I think I need to add time to the timestamps instead of subtract.

retiutut, 2020-01-06:
Fixed!

closed 2020-01-06 by retiutut

retiutut, 2020-01-06:
https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v5.0.0-alpha