OpenBCI/OpenBCI_GUI #418: [USABILITY] Warn users when the dongle and device are not on the same channel

daniellasry, 2019-01-04:
## Problem

It's very confusing for new users if they hit "START SYSTEM" and then "Start Data Stream" but no data is streaming because the dongle and device are not on the same channel (or the device is off!). There is no warning or message to explain why no data is streaming, app behaves as if everything is normal.

When hitting "STATUS", the message "System is down" is pretty vague and does not convey to new users that the device and dongle are not talking at all. Users might assume they are talking, but that something else is wrong with the device.

## Expected 

- When hitting "START SYSTEM" or "Start Data Stream" (ideally both, but at least the latter) warn users that the dongle and the device are not talking (due to channel mismatch, or device being off). A good place for this would be the status bar at the bottom. 

- When hitting "STATUS", we either get "System is up" or "System is down". Replace that with something else that explains that the dongle and the device are either talking or not talking. If they're not talking, suggest the "AUTOSCAN" button.

## Operating System and Version

macOS/Windows/Linux

retiutut, 2019-05-04:
Working on this today!

retiutut, 2019-05-04:
[image]

retiutut, 2019-05-04:
"Success: System is up" or "Failure: System is down" are messages received from the board. As a compromise, the GUI can display additional information as Errors or Successes as seen above.

daniellasry, 2019-05-04:
All we need to do here is replace "system is up" and "system is down" with somethinng more descriptive. For example:

"Success: Cyton dongle and board are connected"

"Failure: Cyton dongle could not connect to the board. Perhaps they are on different channels? Try hitting AUTOSCAN"

Most users will not click "STATUS" before starting the system. They will jsut pick a serial port and click "START SYSTEM". We need a way to show these users the failure error message. What you did with the output() message seems good enough IMO!

retiutut, 2019-05-04:
[image]

retiutut, 2019-05-04:
@daniellasry I can do that. I can catch both of those messages from the board and re-write them.

retiutut, 2019-05-04:
Should be fixed, here is the case where the Cyton board is off but the user selects the Dongle

[image]

daniellasry, 2019-05-04:
Great fix richard!

closed 2019-05-04 by retiutut

retiutut, 2019-05-04:
[image]

PR #508 (2019-05-09): Dev 4.1.2 Minor UI Fixes and Edits
- [x] help dropdown needs to be shifted over

- [x] UI: Fix Help Button and Settings Button Overlap using Button.ignoreHover

- [x] Fix #418
- [x] Fix #493, but also delete invalid files when detected. Currently, it will be remade when user closes or stops GUI, or saves settings.

- [x] Fix #506 and #422 by setting BoardMode using integers

- [x] Fix #509 

____

- [x] update changelog et al and bump version to 4.1.2-beta.2
PR #527 (2019-05-19): Development to master v4.1.2
# v4.1.2
Use OpenBCIHub v2.0.8 please.

## Beta 2

### Improvements
* Add additional button hover text
* Console log message cleanup

### Bug Fixes
* Fix #418, #493, #506, #422, and #509

## Beta 1

### Improvements
* Shorten file names in playback history when needed
* New coloring for Band Power widget
* Smooth and filter dropdowns added to Band Power
* Improved axis labels on Band Power widget
* Cleaned up some console output
* On windows, properly scale for High DPI displays

### Bug Fixes
* Fix crash when opening the console window
* Fix crash when selecting a [... cut]