OpenBCI/OpenBCI_GUI #419: Time Series Widget: Setting window to "7 sec" puts permanent blob on channels 2-7

daniellasry, 2019-01-05:
## Problem

In the time series, change the window to "7 sec".
A big blob appears at the beginning of channels 1-7. The blob stays there until you change the window time again. See pic:

[image]

## Expected 

There should not be a blob in the background when using "7 sec" window

## Operating System and Version

Windows

## GUI Version

4.0.3

## Running standalone app

standalone

## Type of OpenBCI Board

Cyton + Wifi

## Are you using a WiFi Shield?

Yes

retiutut, 2019-01-05:
I already fixed this. It's in the latest pull request.

https://github.com/OpenBCI/OpenBCI_GUI/issues/247

retiutut, 2019-02-14:
@daniellasry I believe this has been fixed as of 4.1.0

closed 2019-02-14 by daniellasry