OpenBCI/OpenBCI_GUI #422: Analog mode '/2', returns random bad values with 4.0.3. But works with 3.3.1

wjcroft, 2019-01-13:
## Problem

See this forum post for background. Skiria is using a photoresistor as external trigger on Aux channel. Everything works fine with 3.3.1 standalone. But fails with latest 4.0.3, 4.0.2.

latest comment
http://openbci.com/forum/index.php?p=/discussion/comment/9958/#Comment_9958
on this thread
http://openbci.com/forum/index.php?p=/discussion/1928/timestamps-to-sync-stimulus-and-eeg-vs-external-hardware-trigger

## Expected 

Aux channel should return photoresistor voltage value. Works fine with 3.3.1.

## Operating System and Version

?? macOS/Windows/Linux

## GUI Version

4.0.3 or 4.0.2 standalone fails. But 3.3.1 works.

## Running standalone app

Standalone. But she also tried 4.0.3 with Processing.

## Type of OpenBCI Board

Cyton firmware 3.1

## Are you using a WiFi Shield?

No

girvent, 2019-01-15:
Windows 8.1 64bit.

brunagirvent, 2019-01-15:
Windows 8.1 64bit

lopezvaa, 2019-05-03:
Hi, I had the same problem with the release version 4.1.2, running on windows 10 Enterprise 64 bit.

I ran the GUI code in Processing. There, the digital read functionallity  works fine and the raw file outputs the digital channels as it is meant to. So, the issue is narrowed down to the stand alone release v4.1.2 (beta) that I downloaded while trying to get my wifishield to work, 

Rolling back to release 4.0.4 fixed it (the latest stable release from the Download section is 4.0.03 also works now)

Kind regards
Alejandro

closed 2019-05-08 by retiutut