OpenBCI/OpenBCI_GUI #425: Read version and copyright date out of plist to avoid duplication in versioning process

produceconsumerobot, 2019-01-19:
## Problem

Explain the problem

## Expected 

In a perfect world, what do you expect to happen.

## Operating System and Version

macOS/Windows/Linux

## GUI Version

The version is displayed on startup.

## Running standalone app

Are you running the downloaded app or are you running from Processing 3

## Type of OpenBCI Board

Cyton/Cyton+Daisy/Ganglion

## Are you using a WiFi Shield?

Yes/No

retiutut, 2019-01-29:
I did try this, and doesn't really work for the Info.plist file in the Mac app package. I did get it to work from the Info.plist.tpml file during dev with. Bottom line, it's much easier the way it is. Simply update OpenBCI_GUI.pde and the plist.

retiutut, 2019-03-04:
the file becomes info.plist in the standalone

and in a parent directory instead of same
that's the trick it seams

daniellasry, 2019-04-21:
all the possible solutions are more complex and less desirable than the current situation. not fixing.

closed 2019-04-21 by daniellasry