OpenBCI/OpenBCI_GUI #426: GUI sometimes freezes at startup with a blank screen

daniellasry, 2019-01-21:
## Problem

Occasionally, the GUI will start up with a blank/grey screen and stay stuck that way, requiring a relaunch.

It's a little difficult to reproduce. I have done it on windows, but it's easily reproducible on an older Mac while simultaneously recording the screen and having a skype conference. The slower the system, the easier it is to reproduce, since it's a timeout issue.

When reproducing from processing, here is the error message:

```
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Mac
Welcome to the Processing-based OpenBCI GUI!
For more information about how to work with this code base, please visit: http://docs.openbci.com/OpenBCI%20Software/
Local Version: 403, Latest Version: 402
GUI is up to date!
Graphics & GUI Library: ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
java.lang.RuntimeException: Waited 5000ms for: <17009c9e, 19c7700e>[count 2, qsz 1, owner <main-FPSAWTAnimator#00-Timer0>] - <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1>
        at processing.opengl.PSurfaceJOGL$2.run(PSurfaceJOGL.java:410)
        at java.lang.Thread.run(Thread.java:748)
```

## Operating System and Version

macOS/Windows/Linux

## GUI Version

v4.0.3 (and earlier ones too)

## Running standalone app

Standalone and Processing

retiutut, 2019-01-21:
@daniellasry All of what you have written is accurate. I have experience this when launching the GUI for the first time after a login on my Mac.

redFrik, 2019-02-01:
i found a way to reliably reproduce this grey-screen-lock at startup by using the network filter software called [little snitch](https://www.obdev.at/products/littlesnitch/index.html). 
directly after launching OpenBCI_GUI.app the network filter will pop up a dialog asking if the app may connect to github.  it doesn't matter what i select - deny or allow - the app always locks with a blank screen.  if i deactivate littlesnitch, or select 'forever' so that the filter dialog doesn't show, the OpenBCI_GUI.app starts as expected.

[image]

(OpenBCI_GUI.app 4.0.4, macOS 10.12.6)

also confirmed the same problem with Processing 3.5.2 and latest master...

```
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Mac
Welcome to the Processing-based OpenBCI GUI!
For more information about how to work with this code base, please visit: http://docs.openbci.com/OpenBCI%20Software/
RunnableTask.run(): A caught exception occured on thread main-Display-.macosx_nil-1-EDT-1: RunnableTask[enqueued true[executed false, flushed false], tTotal 0 ms, tExec 0 ms, tQueue 0 ms, attachment null, throwable java.lang.RuntimeException: Waited 5000ms for: <1665e9c8, 7a1737e3>[count 2, qsz 1, owner <main-FPSAWTAnimator#00-Timer0>] - <main-Display-.macosx_nil-1-EDT-1>]
java.lang.RuntimeException: Waited 5000ms for: <1665e9c8, 7a1737e3>[count 2, qsz 0, owner <main-FPSAWTAnimator#00-Timer0>] - <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1>
	at processing.opengl.PSurfaceJOGL$2.run(PSurfaceJOGL.java:410)
	at java.lang.Thread.run(Thread.java:748)
RuntimeException: Waited 5000ms for: <1665e9c8, 7a1737e3>[count 2, qsz 0, owner <main-FPSAWTAnimator#00-Timer0>] - <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1>
```

retiutut, 2019-02-01:
Putting a network filter on the app will cause a timeout when the GUI tries to read HTML text from GitHub latest release page. I don’t recommend this.

redFrik, 2019-02-01:
yes, and this is the whole idea.  to force a timeout in a reliable way and help the developers track down the problem.  imho, the app should timeout gracefully and not lock up.

daniellasry, 2019-02-02:
I'm testing a fix for this now.

commit 6094a3c5 (2019-02-09): Merge pull request #433 from daniellasry/fix-timeout

[FIX] the GUI timing out on launch. fix #409 fix #406 fix #426
commit 661d74ee (2019-02-09): [FIX] the GUI timing out on launch. fix #409 fix #406 fix #426
closed 2019-03-09 by daniellasry