OpenBCI/OpenBCI_GUI #438: Start data stream should clear previous time series and accelerometer screen

jps2000, 2019-02-11:
## Problem

Start data stream  does not clear time series and accelerometer

## Expected 
start always with  blank screens. If there is remaining data this is confusing.

## Operating System and Version
w7

## GUI Version

4.0.1

## Running standalone app

downloaded app 

## Type of OpenBCI Board

/Ganglion

## Are you using a WiFi Shield?

No

retiutut, 2019-02-14:
@jps2000 Is this after you press Stop System and then press Start System?

jps2000, 2019-02-16:
No;  Start system clears the time series + acc already.
I am talking about **Start Data Stream**.  Clear traces is in particular  important  in  playback when moving to another time and start data stream  again.
Try  following sequence:
do playback of a file
press start data stream
press stop data stream
move slider to  a different time point
press start data stream to continue 
then you see old data and new data  moving along the time series and acc without a gap

But also during normal mode it is strange behaviour to see recordings in one sequence coming from different times. So I suggest to clear times series and accelerometer whenever start data stream is pressed.

retiutut, 2019-02-22:
@jps2000 Thank you for describing the process step by step!

I think the crucial step is when you *move slider to a different time point*. Currently, accelerometer and time series are cleared when we use `skipToStartButtonAction()`. **I'm sure I can apply this to when the slider is moved.**

getting input from @produceconsumerobot

retiutut, 2019-02-22:
@produceconsumerobot Agrees with you @jps2000. 😄 

I will try to implement the following fixes:

- Clear accelerometer and time series when playback slider is moved

- Clear accelerometer and time series when `start data stream` is pressed

jps2000, 2019-03-20:
Thank you looks, good now

PR #444 (2019-03-10): Development v4.1.1-beta.0
- [x] Fix issue with Playback Range Select by updating the Array to a StringList

- [x] Fixes #443 

- [x] Fixes #438

- [x] Clear accelerometer and time series when playback slider is moved

- [x] Clear accelerometer and time series when `start data stream` is pressed in playback mode

____

- [x] Fix Bug: Load FFT X and Y axis default settings properly for all data modes https://github.com/OpenBCI/OpenBCI_GUI/pull/444/commits/26a96f4a34a30ddc451b27447d3d0ed596dac22b
- [x] Fix Bug: Accel Y auto scale functionality https://github.com/OpenBCI/OpenBCI_GUI/pull/444/commi [... cut]
closed 2019-03-20 by jps2000