OpenBCI/OpenBCI_GUI #439: Add a "console" window that displays all output, for easier troubleshooting

daniellasry, 2019-02-11:
We get a lot of issues like:

- I hit "START STREAMING" and nothing happens
- The GUI freezes on startup
- etc.

Sure, it would be great to have more robust feedback to the user when something fails.

But in the meantime, it would be helpful to have a way for users to look at the program's console output.

retiutut, 2019-02-11:
Proof of Concept: 

```
import java.io.PrintStream;
import java.io.FileOutputStream;

int myTimer;
PrintStream original = new PrintStream(System.out);

void setup() {

  println("This goes to the console.");
  try {
    String file = dataPath("")+"console.txt";
    if (!new File(dataPath("")).isDirectory()) {
      if (!new File(dataPath("")).mkdirs()) {
        System.err.println("Directory creation failed!");
        exit();
      }
    }
    FileOutputStream outStr = new FileOutputStream(file, false);
    PrintStream printStream = new PrintStream(outStr);

    System.setOut(printStream);
    System.setErr(printStream);
  }
  catch (IOException e) {
    System.err.println("Error! Check path, or filename, or security manager! "+e);
    exit();
  }
  println("This goes to the file");
  println("Hello major tom!");

}

void draw() {
  if (millis() > myTimer + 2000) {
    println("current time: " + millis());
    String[] temp = loadStrings("dataconsole.txt");
    original.println(temp[temp.length-1]);
    myTimer = millis();
  }

}
```

retiutut, 2019-02-23:
https://gist.github.com/atduskgreg/666e46c8408e2a33b09a

retiutut, 2019-02-24:
There is a problem when scaling if you keep reloading the entire file as it expands with
```
String[] temp = loadStrings("dataconsole.txt");
    original.println(temp[temp.length-1]);
```

The solution is a new method to print to both file and console in a simple way:
```
void consolePrint(String _output) {
  println(_output);
  original.println(_output);
}
```

retiutut, 2019-02-24:
```
import java.io.PrintStream;
import java.io.FileOutputStream;
import java.util.Arrays;

int myTimer;

ScrollRect scrollRect;        // the vertical scroll bar
float heightOfCanvas = 500;  // realHeight of the entire scene

PrintStream original = new PrintStream(System.out);
ConsoleData consoleData = new ConsoleData();

void setup() {
  size(500,500);
  println("This goes to the console.");
  consoleData.setupConsoleOutput();
  consolePrint("This goes to the file and the Console");
  consolePrint("Hello major tom!");

  scrollRect = new ScrollRect();
  background(90);
}

void draw() {
  clear();

  if (millis() > myTimer + 100) {
    consolePrint(" " + ++consoleData.outputLine);
    myTimer = millis();
  }

  fill(255);

  scene();
  scrollRect.display();
  scrollRect.update();

  /*
  for (int i = 0; i < consoleData.data.size(); i++) {
    text(consoleData.data.get(i), 10, 10 * i + 3, 255, 80);
  }
  */
}

void mousePressed() {
  scrollRect.mousePressedRect();
}

void mouseReleased() {
  scrollRect.mouseReleasedRect();
}

void consolePrint(String _output) {
  println(_output);
  original.println(_output);
  consoleData.data.append(_output);
}

class ConsoleData {

  StringList data = new StringList();
  int outputLine = 0;

  void setupConsoleOutput() {
    try {
      String file = dataPath("") + "Console.txt";
      FileOutputStream outStr = new FileOutputStream(file, false);
      PrintStream printStream = new PrintStream(outStr);
      System.setOut(printStream);
      System.setErr(printStream);
    }
    catch (IOException e) {
      System.err.println("Error! Check path, or filename, or security manager! "+e);
    }
  }

}

// --------------------------------------------------------------

void scene() {
  pushMatrix();

  int fontHeight = 12;
  // reading scroll bar
  float newYValue = scrollRect.scrollValue();
  translate (0, newYValue);

  if ((fontHeight*(consoleData.data.size() - 1) + 4) > heightOfCanvas) {
    heightOfCanvas += 80;
  }

  for (int i = 0; i < consoleData.data.size(); i++) {
    text(consoleData.data.get(i), 10, fontHeight * i + 4, 255, 80);
  }

  text("End of virtual canvas", 380, heightOfCanvas-16);
  fill(122);
  popMatrix();
}

// ===============================================================

class ScrollRect {

  float rectPosX=0;
  float rectPosY=0;
  float rectWidth=14;
  float rectHeight=30;

  boolean holdScrollRect=false;

  float offsetMouseY;

  //constr
  ScrollRect() {
    // you have to make a scrollRect in setup after size()
    rectPosX=width-rectWidth-1;
  }//constr

  void display() {
    fill(122);
    stroke(0);
    line (rectPosX-1, 0,
      rectPosX-1, height);
    rect(rectPosX, rectPosY,
      rectWidth, rectHeight);

    // Three small lines in the center
    centerLine(-3);
    centerLine(0);
    centerLine(3);
  }

  void centerLine(float offset) {
    line(rectPosX+3, rectPosY+rectHeight/2+offset,
      rectPosX+rectWidth-3, rectPosY+rectHeight/2+offset);
  }

  void mousePressedRect() {
    if (mouseOver()) {
      holdScrollRect=true;
      offsetMouseY=mouseY-rectPosY;
    }
  }

  void mouseReleasedRect() {
    scrollRect.holdScrollRect=false;
  }

  void update() {
    // dragging of the mouse
    if (holdScrollRect) {
      rectPosY=mouseY-offsetMouseY;
      if (rectPosY<0)
        rectPosY=0;
      if (rectPosY+rectHeight>height-1)
        rectPosY=height-rectHeight-1;
    }
  }

  float scrollValue() {
    return
      map(rectPosY,
      0, height-rectHeight,
      0, - (heightOfCanvas - height));
  }

  boolean mouseOver() {
    return mouseX>rectPosX&&
      mouseX<rectPosX+rectWidth&&
      mouseY>rectPosY&&
      mouseY<rectPosY+rectHeight;
  }//function
  //
}//class
```

retiutut, 2019-02-24:
[image]

retiutut, 2019-02-24:
https://github.com/retiutut/OpenBCI_GUI/blob/consoleWindowGUI/Feature_Tests/consoleLogTest/consoleLogTest.pde

retiutut, 2019-02-24:
This goes to the file and the console.
Hello Major Tom!
1
2
3
4
5
6
7
8
9
10
11
12
13
14

retiutut, 2019-02-24:
```
This goes to the file and the console.
Hello Major Tom!
1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
22
23
24
25
26
27
28
29
30
31
32
33
34
35
36
37
38
39
Copying console log to clipboard!
```

retiutut, 2019-02-24:
[image]

retiutut, 2019-02-24:
[image]
```
This goes to the file and the console.
Hello Major Tom!
Copying console log to clipboard!
```

PR #448 (2019-03-13): Add Feature: Console Log Window
Fixes #439
- [x] Implement front-end solution (Cog Button in GUI top right)
- [x] Implement consoleLogTest.pde with at-scale solution
- [x] Test scrolling functionality
- [x] Discuss when consoleWindow should be redrawn so it doesn't impact overall GUI performance. 
*Update occurs based on user interaction and consolePrint()s only* 😎 
_____
- [x] Console window appears behind the gui. Should appear in front. https://github.com/OpenBCI/OpenBCI_GUI/pull/448/commits/ff4dcb059556ec4f89df1263e897d12f2673304e
- [x] Pressing the button multiple times opens multiple con [... cut]
commit 61469f64 (2019-03-13): Add Feature: Console Log Window for the GUI Fixes #439

Still needs work, but it builds and runs!
closed 2019-03-13 by daniellasry