OpenBCI/OpenBCI_GUI #440: OpenBCI GUI (4.0.4 & 4.1.0-beta0) freezes on CH16 impedance check button

miladinovic, 2019-02-14:
## Problem

The openbci gui freezes when I try to check the impendance on the channel 16. By clciking on the "ohm" icom everthing stops, as shown in the video--> https://www.youtube.com/watch?v=ztb_Z7bxpSw&feature=youtu.be

In processing i get exception array out of bounds in the file HardwareSettingsController line 534

## Expected 
To get impedance value from ch 16.

## Operating System and Version

macOS/Windows (tried on both)

## GUI Version

4.0.4 and 4.1.0-beta.0

## Running standalone app

Running standalone app & Processing

## Type of OpenBCI Board

Cyton+Daisy

## Are you using a WiFi Shield?

No

retiutut, 2019-02-17:
I thought I accounted for this in one of the other recent patches. I'll look into this. Should be a quick fix.

retiutut, 2019-02-17:
Fixed 🔥 🖥 
[image]

retiutut, 2019-02-22:
@daniellasry This has been fixed!

commit f07fb8ad (2019-02-21): Fix Cyton Impedance Check, #Fixes #440
closed 2019-02-26 by daniellasry