OpenBCI/OpenBCI_GUI #443: Time in status line is not updated every seconds

jps2000, 2019-02-19:
## Problem
The status line shows absolute time and relative time in seconds in playback
In playback of  longer recordings ( 20 min or so) the absolute time is not updated every second. However the relative time changes every second correctly.

## Expected 
Update  absolute time also every second

## Operating System and Version
Win

## GUI Version

4.1.0 beta

## Running standalone app

 downloaded app 

## Type of OpenBCI Board

Ganglion

## Are you using a WiFi Shield?

No

retiutut, 2019-02-22:
@jps2000 Making a 20+ minute recording now to test this...

retiutut, 2019-02-22:
@jps2000 I have reproduced this issue!

[image]

retiutut, 2019-02-22:
`findTimesToDisplay()` in W_TimeSeries.pde can probably be updated to use the actual timestamps that appear!

retiutut, 2019-02-22:
I'm going to try pulling from the regular time column:
```
127, 0.00, 0.00, 0.00, 0.00, 0.000, 0.000, 0.000, 10:05:14.214, 1550851514214
0, 17.02, 14.61, 8.28, 23.24, 0.000, 0.000, 0.000, 10:05:14.227, 1550851514227
```

retiutut, 2019-02-22:
Fixed! 🔥 💻 
[image]

@jps2000 Do you think we should show the milliseconds?

jps2000, 2019-02-22:
Absolute time is important to trace back events with videos for example. Hence there is no need for milliseconds as long as there is no slow motion  playback . 
But when you have ms now then leave it.

@retiutut 
By the way: Do you see any chance to speed up loading/handling   of long playback files?
For example sleep studies etc need analysis of recordings  in the magnitude of hours.
Such long files can not be handled now
An "easy to realize" (nothing is easy) remedy may be  a recording scheduler to  limit length of recordings
( start new recording after x minutes)

retiutut, 2019-02-22:
Ok leaving milliseconds in. Was able to update the elapsed time algorithm also to be very accurate.

@jps2000 Let's keep #400 by itself. The solution to that issue should resolve any problems users should have resulting from working with long playback files.

retiutut, 2019-03-18:
@jps2000 I applied a fix to for this issue in a recent update. I think this issue can be closed.

PR #444 (2019-03-10): Development v4.1.1-beta.0
- [x] Fix issue with Playback Range Select by updating the Array to a StringList

- [x] Fixes #443 

- [x] Fixes #438

- [x] Clear accelerometer and time series when playback slider is moved

- [x] Clear accelerometer and time series when `start data stream` is pressed in playback mode

____

- [x] Fix Bug: Load FFT X and Y axis default settings properly for all data modes https://github.com/OpenBCI/OpenBCI_GUI/pull/444/commits/26a96f4a34a30ddc451b27447d3d0ed596dac22b
- [x] Fix Bug: Accel Y auto scale functionality https://github.com/OpenBCI/OpenBCI_GUI/pull/444/commi [... cut]
commit 10e1962b (2019-03-10): Fix Bug: Playback absoulute time not updated every second #closes #443

Now displaying milliseconds!!! Huge improvement with less code. This is 
possible because of updated timestamps in playback files.
closed 2019-03-20 by jps2000