OpenBCI/OpenBCI_GUI #445: Connection Issues and program freeze

wildsoil, 2019-02-22:
## Problem

Explain the problem
Ive had this setup for about a month. The first week connection was simple. But about 2 weeks back Ithings just stopped connecting. After updating to 4.1 it will connect on the Mac but only after scanning several times but when I push the start button the program freezes. It even freezes when running synthetic input. In Neuromore on Mac it no longer recognizes the cyton stating "no connection" however when I search the ports the dongle shows up in port 3.  In windows on an old lenovo laptop running W7pro the GUI no longer recognizes the dongle in port 3 even after several scans. 

## Expected
 It did connect for about two weeks on the mac in neuromore but not GUI and did connect on windows in both GUI and Neuromore. 

In a perfect world, what do you expect to happen.
Turn it off then turn it on and everything is all ducky
## Operating System and Version
2012 Mac Mini 3gig 16gb Mojave 
Lenova Laptop intel I7 2.9ghz 8gb Windows 7 pro

macOS/Windows/Linux Mac Mojave 

## GUI Version
4.1 
The version is displayed on startup.
4.1
## Running standalone app

Are you running the downloaded app or are you running from Processing 3
Open GUI and Neuromore
## Type of OpenBCI Board

Cyton/Daisy

## Are you using a WiFi Shield?

No just the BT dongle

retiutut, 2019-02-22:
The GUI has improved significantly from 3.0-4.1. Coding doesn't degrade over time, so that means there is something wrong with your computers, authenticity of OpenBCI tech, or the firmware.

Also, this is not a place for Neuromore support. This is a repository for the OpenBCI GUI.

daniellasry, 2019-02-26:
Hey @wildsoil ,

As mentioned above, try to run the GUI from processing: http://docs.openbci.com/OpenBCI%20Software/01-OpenBCI_GUI#the-openbci-gui-running-the-openbci-gui-from-the-processing-ide

Hopefully, the console output in Processing can help us diagnose the issue. I'm not sure what would cause the program to freeze, especially if it worked before.

wildsoil, 2019-02-27:
Thanks much Daniel,  I started the Processing install last night but was
unable to get any files into the program. I did rename the gui master also.
BTW the GUI has always frozen on the mac when starting even with synthetic.
I also started VM ware w7 pro virtual on the mac but am having screen size
issues (left side of screen is cut off). Thanks for your help and patience!

Trevor

daniellasry, 2019-03-30:
Hey @wildsoil , check out [v4.1.1-beta.0 of the GUI](https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v4.1.1-beta.0)

We added a console window (little button on the upper right) where you can see debug output from the GUI. The same output is written to SavedData/Settings/console-data.txt

Could you share the output of console-data.txt when the program crashes? Thanks!

wildsoil, 2019-03-31:
Dan, you rock dude! Thanks for this!  the only problem is that I can't get
it to crash ;) The new hub and 4.1.1 beta has this thing connecting quickly
and working solid on my mac. Haven't tried on the PC yet.  I can't tell you
how many hours I have into troubleshooting,  my wife kept yelling at me to
return it ;)  You saved me!  I love the console window idea too.
Best,   Trevor

daniellasry, 2019-03-31:
That's great news! Hopefully, if it ever happens again we will have an easy way to diagnose it.

Credits to @retiutut for implementing the console window 🥇

daniellasry, 2019-04-19:
I believe I've fixed this issue in the 4.1.2 beta release. Be sure to let me know if this happens again

closed 2019-04-19 by daniellasry