OpenBCI/OpenBCI_GUI #446: Consider whether OpenBCI GUI should launch with streaming data (rather than requiring another step)

produceconsumerobot, 2019-02-22:


retiutut, 2019-03-26:
@jps2000 What do you think about this?

Should the GUI start streaming data after you press "Start System"?

jps2000, 2019-03-27:
It is good custom for example in ecg that once you connect everything and switch it on that on see traces . Then there is the possibility to correct or readjust electrodes, check impedances etc.
If everything is fine ad the patieont is quiet then one can press a recording button to log incoming data.
 Keeping this in mind I suggest the following
after connect show time series and accelerometer ( default on)
there is a button to start stop logging always beginning with a new file name coded by date and time. (preferably start time)
There is a possibility to limit length of recordings by a preset duration.
So once logging has started, a timer  looks for duration of the recording. 
If preset time is over it stops logging, close the file  and start new recording with new file name.

closed 2019-05-07 by retiutut