OpenBCI/OpenBCI_GUI #447: OpenBCI_Hub 2.0.6 not initialize in the OpenBCI_GUI 4.1.0 using Processing.

wellton, 2019-02-25:
## Problem

The OpenBCI_Hub 2.0.6 not initialize in the OpenBCI_GUI 4.1.0 using Processing.

IMPORTANT = The OpenBCI_GUI version 4.0.4 with OpenBCI_Hub 2.0.5 Works, using same procedures used in the 4.1.0

obs. I put the Hub inside folder data.

[image]

[image]

## Expected 

Do initialize the Hub.

## Operating System and Version

Debian GNU/Linux Buster

## GUI Version

4.1.0

## Running standalone app

Are you running the downloaded app or are you running from Processing 3

## Type of OpenBCI Board

Cyton+Daisy

## Are you using a WiFi Shield?

Yes

retiutut, 2019-02-25:
This is likely only happening on Linux.

daniellasry, 2019-02-26:
Ah. I believe this is due to a change I made. As of 4.1.0, you need to put the hub inside a "OpenBCIHub" folder within the "data" folder. Let us know if that fixes the issue.

wellton, 2019-02-26:
@daniellasry Hi Daniel. Yes. Its Works. =) Thank you.

[image]

[image]

[image]

pranavsuresh7, 2019-02-28:
Hey I also face the same problem mine ubuntu version is 18.04.
[image]
[image]

pranavsuresh7, 2019-02-28:
Fixed it  by   sudo apt install libgconf-2-4 ,installing this library and sudo reboot. It was issue with OpenBCI_HUB

retiutut, 2019-02-28:
@pranavsuresh7 You are using an older version of the GUI. Please try [the latest release](https://github.com/OpenBCI/OpenBCI_GUI/releases/latest)

closed 2019-03-04 by daniellasry