OpenBCI/OpenBCI_GUI #458: GUI hangs even when 'Synthetic (algorithmic)' is chosen as the data source.

shirleyzhang867, 2019-03-11:
## Problem
Posting on behalf of customer-
CytonDaisy user with functional hardware is having issues with GUI v4.0.3 and v4.1.0. Software hangs when 'start data-acquisition' is selected.
"The GUI opens, the status indicates 'success', but when we press the 'start data-acquisition' button, nothing happens. The system just hangs."
"The software hangs even when I choose 'Synthetic (algorithmic)' as the data source."
"the hub says it is v.2.0.6"

## Operating System and Version
CytonDaisy
MacOs 10.13.1 High Sierra

## GUI Version
 4.0.3 and 4.1.0

retiutut, 2019-03-11:
There should be a message displaying at the bottom of the GUI as to why.

retiutut, 2019-03-11:
Is this a duplicate issue of https://github.com/OpenBCI/OpenBCI_GUI/issues/445 ?

retiutut, 2019-03-12:
Based on the information provided:

- GUI standalone needs more testing to determine what could be happening
- ConsoleLog feature may provide more insight

shirleyzhang867, 2019-03-19:
Running the GUI from Processing fixed the GUI hanging issue.

closed 2019-03-19 by shirleyzhang867