OpenBCI/OpenBCI_GUI #461: Add support for longer recordings in GUI

jps2000, 2019-03-19:
## Problem

Playback or handling long recordings is a problem due to file size. However, sleep studies require long recording and it is laborious  and erroneous to start and stop recordings  every 20-30 minutes.

## Expected 
There could be a recording scheduler to  start a new file every xx minutes
in the Pick Transfer Protocol box there is the DATA LOG FILE section where a file name can be entered. There may be added a line  Log time limit    (no, 10, 20, 40, 60 minutes)
Data logging should end after this time and restarted asap with a new file name.
This prevents  from getting  too long recordings and makes the system more robust for data loss.

## Operating System and Version

W

## GUI Version

all

## Running standalone app

downloaded app 

## Type of OpenBCI Board

Cyton/Cyton+Daisy/Ganglion

## Are you using a WiFi Shield?

No

retiutut, 2019-03-26:
@jps2000 Thank you for making this issue! It is on the to-do list!

retiutut, 2019-04-14:
@jps2000 Check out https://github.com/OpenBCI/OpenBCI_GUI/issues/483 ! This should work!

retiutut, 2019-04-28:
https://openbci.com/index.php/forum/#/discussion/1668/unable-to-convert-large-files-from-sd-card-out-of-memory-error

jps2000, 2019-05-30:
Thank you
But still:  splitting long events into digestable pieces is a good one. It also prevents inherently (huge) data loss.

retiutut, 2019-06-21:
[image]

retiutut, 2019-06-22:
[image]

jps2000, 2019-06-22:
Looks great
thank you. 
Small comments: call it Max file  duration or length instead of size 
length:
infinite, 5, 15, 30, 60,120 as length
just to get it right:
after time over a new recording is started and ends again after preset minutes.

retiutut, 2019-06-22:
I'm **not** going to put a "No Limit" or "Infinite" option, as this defeats the point of this fix.

I **can** change the options to 5,15,30,60,90,120.

retiutut, 2019-06-22:
I've changed my mind. Let's do:

5, 15, 30, 60, 120, No Limit

retiutut, 2019-06-22:
[image]

retiutut, 2019-06-22:
Super excited to share this next picture!!! Zero data loss using nanoTime()!!! Files set to 1 min max length produce files of exact same size.
[image]

retiutut, 2019-06-22:
Done 🔥 🔥 🔥 
[image]

PR #547 (2019-07-01): Restructure GUI and Recording Data in User's Documents Folder
Fixes #461 and #483 

**Noticeable Changes**
- [x] Data generated by the GUI is neatly organized in the User's Documents folder now
- [x] OpenBCI Data Format: Pausing data stream starts new recording
- [x] OpenBCI Data Format: Impose max file size, auto closes/opens files and generates file name based on date and time
- [x] BDF Data Format: temp file is created and destroyed in the User's Documents folder
- [x] BDF Data Format: All BDF session files appear in OpenBCI_GUI/Recordings next to OpenBCI Data format sessions

____

- [x] U [... cut]
closed 2019-06-22 by retiutut