OpenBCI/OpenBCI_GUI #462: Improve band power widget

jps2000, 2019-03-19:
## Problem
The chart would look better if the bars are colored  
blue, magenta, green, yellow red  (left to right)
The response time goes with smooth of the FFT plot ??
## Expected 

Color the bars blue, magenta, green, yellow red  (left to right)
Provide dedicated smoothing adjustment independent of FFT

## Operating System and Version

w

## GUI Version
all

## Running standalone app

app

## Type of OpenBCI Board

Cyton/Cyton+Daisy/Ganglion

## Are you using a WiFi Shield?

n

retiutut, 2019-03-26:
@jps2000 Adding this issue to the to-do list also!

retiutut, 2019-05-04:
[image]

closed 2019-05-04 by retiutut