OpenBCI/OpenBCI_GUI #463: OSC data stream not working

jzingh98, 2019-03-24:
## Problem
I'm using the OpenBCI HUB and GUI with a Ganglion board. Running it as administrator. I'm trying to get it to send OSC data to neuromore but it was unable to detect any OSC at the specified port and address. Also downloaded the Kasper Kaperman OSC data viewers, and it showed no incoming data stream at the specified port and address.

## Expected 

In a perfect world, what do you expect to happen.

## Operating System and Version
Windows 64 bit

## GUI Version
4.0.3

## Running standalone app

Are you running the downloaded app or are you running from Processing 3

## Type of OpenBCI Board

Ganglion

## Are you using a WiFi Shield?

No

retiutut, 2019-03-26:
I was not able to replicate this issue. [OSC](https://en.wikipedia.org/wiki/Open_Sound_Control) is working perfectly, and has been for as long as I have used the GUI.

Tested using Windows 10, GUI 4.1.0, and https://www.kasperkamperman.com/blog/processing-code/osc-datamonitor/

[image]

>  Also downloaded the Kasper Kaperman OSC data viewers, and it showed no incoming data stream at the specified port and address.

This would only happen if the port an address are incorrect. All you need to get OSC working is the correct IP and port, as seen in the above screenshot.

This issue can be closed.

jzingh98, 2019-03-26:
I think you might be right about it being a port or IP issue. Not working on OpenVibe either.

It's a Ganglion board, Windows 10, and currently using latest version of each Hub, GUI, and OpenVibe

[image]

retiutut, 2019-03-26:
That's not how you connect the GUI to OpenVibe. That driver you have selected in OpenVibe is for the Cyton only, and does not require the GUI.

jzingh98, 2019-03-26:
Okay that makes sense, thank you for clarifying.  What did you mean by Run the GUI from Processing?

retiutut, 2019-03-26:
@jzingh98 This is a place for issues with GUI, not a place for step-by-step instructions. For this reason, there is a wealth of information in the Docs. https://docs.openbci.com/OpenBCI%20Software/01-OpenBCI_GUI#the-openbci-gui-running-the-openbci-gui-from-the-processing-ide

There is no problem with the GUI related to the issue of "OSC data stream not working" because it works just fine. **This issue can be closed.**

closed 2019-03-26 by jzingh98

retiutut, 2019-03-26:
@jzingh98 Once you find a setup that you think will work for what you want to do, work hard for a day or two to make sure you've done everything based on what you can read.

Example: I like to run the GUI and stream via OSC to [Max/MSP](https://en.wikipedia.org/wiki/Max_(software)) to control sounds/lights/etc.

It seems like researchers prefer LSL.

If something is legitimately broken, let us know and we will get to the bottom of it!

retiutut, 2019-03-26:
@jzingh98 If you really want to use OpenVibe, I can help setup the example file!

retiutut, 2019-03-26:
@jzingh98 Here you go! Going to provide a short how to right now since it doesn't seem to be in the Docs:

1. Startup GUI from Processing and connect to Ganglion
2. Select LSL in the Networking Widget (see screenshot)
[image]

3.