OpenBCI/OpenBCI_GUI #481: problem to run openbci cyton

babilord, 2019-04-14:
## Problem

hi, i have been using cyton openbci since 2 years. everything was good and openbci code worked perfectly, but 2 days I can't work with openbci, because when I click on start then I see this error : java.lang.pointerException. I don't change my code!! what happen?! so, you can see this error in the this picture. please help me, thank u...

## Expected 

In a perfect world, what do you expect to happen.

## Operating System and Version

Windows

## GUI Version

processing 2.2.1

## Running standalone app
processing 2.2.1

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

retiutut, 2019-04-15:
What version of the OpenBCI_GUI are you using?

babilord, 2019-04-15:
I don't know :| how can I understand it?

daniellasry, 2019-04-15:
What do you see here?

[image]

closed 2019-06-10 by retiutut