OpenBCI/OpenBCI_GUI #483: Organize recording sessions in their own timestamped folder

daniellasry, 2019-04-14:
Proposal for new folder structure:

SavedData/
|---- console-data.txt
|---- Sample_Data/
|---- Settings/
|---- Recordings/
        |---- OpenBCISession-2019-04-14_20-40-48/
                |---- OpenBCI-RAW-2019-04-14_20-40-48.txt
                |---- OpenBCI-RAW-2019-04-14_20-45-32.txt
                |---- OpenBCI-RAW-2019-04-14_20-56-08.txt
                |---- OpenBCI-RAW-2019-04-14_21-03-23.txt
        |---- OpenBCISession_2019-04-15_09-40-48/
                |---- OpenBCI-RAW-2019-04-15_09-40-48.txt
                |---- OpenBCI-RAW-2019-04-15_09-44-21.txt
                |---- OpenBCI-RAW-2019-04-15_09-59-03.txt
                |---- OpenBCI-RAW-2019-04-15_10-11-43.txt
        |---- etc...
                |---- etc...
                |---- etc...
        |---- etc...
                |---- etc...
                |---- etc...

retiutut, 2019-04-14:
This includes moving console-data.txt from SavedData/Settings/ to SavedData.

retiutut, 2019-04-14:
Above proposed solution Fixes #461 🔥

retiutut, 2019-06-18:
Requirement: Change name of “Saved File” to “Saved Session”

retiutut, 2019-06-21:
Requirement: create `Documents/OpenBCI_GU/Sample_DataI` and copy sample data on first launch, or if the files don't exist

retiutut, 2019-06-22:
Requirement: Console Data will have to be in its own folder, because this describes activity from App Start to App Stop, as opposed to Session Start/Stop

retiutut, 2019-06-22:
C'est très bon!
[image]

retiutut, 2019-06-22:
Done 🔥 🔥 🔥 
[image]

closed 2019-06-22 by retiutut

daniellasry, 2019-06-24:
Nailed it!