OpenBCI/OpenBCI_GUI #489: GUI unable to connect to OpenBCI HUB

JayroMartinez, 2019-04-18:
## Problem

I'm using the OpenBCI GUI with Processing on a Raspberry Pi 3. When I run the GUI and select my board (Cython) the "Pick Transfer Protocol" menu doesn't appear. Instead I have  a "Hub not connected" menu with an "Open BCI install guide" button. I've followed the instructions here:

[https://docs.openbci.com/OpenBCI%20Software/01-OpenBCI_GUI]

So OpenBCIHub folder is in /OpenBCI_GUI/data folder.

## Expected 

In a perfect world, the Pick TRansfer Protocol menu should appear.

## Operating System and Version

Raspbian GNU/Linux 9 (stretch)

## GUI Version

4.1.2

## Running standalone app

Processing 3.5.3

## Type of OpenBCI Board

Cyton+Daisy

## Are you using a WiFi Shield?

No

## Processing Console Log

Screen Resolution: 1824 X 984
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://docs.openbci.com/OpenBCI%20Software/
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Linux
Local Version: 412, Latest Version: 412
GUI is up to date!
attempting to load images for presentation...
DONE loading images!
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
The file "SavedData/Settings/UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
java.net.ConnectException: Connection refused (Connection refused)
	at java.net.PlainSocketImpl.socketConnect(Native Method)
	at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:350)
	at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at java.net.Socket.connect(Socket.java:538)
	at java.net.Socket.<init>(Socket.java:434)
	at java.net.Socket.<init>(Socket.java:211)
	at processing.net.Client.<init>(Unknown Source)
	at OpenBCI_GUI$Hub.startTCPClient(OpenBCI_GUI.java:10551)
	at OpenBCI_GUI$Hub.<init>(OpenBCI_GUI.java:10527)
	at OpenBCI_GUI.systemUpdate(OpenBCI_GUI.java:1107)
	at OpenBCI_GUI.draw(OpenBCI_GUI.java:544)
	at processing.core.PApplet.handleDraw(PApplet.java:2475)
	at processing.opengl.PSurfaceJOGL$DrawListener.display(PSurfaceJOGL.java:866)
	at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
	at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
	at jogamp.opengl.GLAutoDrawableBase$2.run(GLAutoDrawableBase.java:443)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
	at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
	at com.jogamp.newt.opengl.GLWindow.display(GLWindow.java:759)
	at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:81)
	at com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:452)
	at com.jogamp.opengl.util.FPSAnimator$MainTask.run(FPSAnimator.java:178)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
[WARN]: Failed to connect to OpenBCIHub background application. LIVE functionality will be disabled.
Instantiating hub object...

daniellasry, 2019-04-18:
I appreciate all the details and the console log output!
We don't test our software on raspberry Pi, but there's still a chance it can work. My guess here is that the HUB application is not compiled for ARM, and therefore refuses to launch.

To be sure, you can try launching the HUB on its own and verify that it's running: 
[image]

To launch the hub, run the executable: `OpenBCI_GUI/data/OpenBCIHub/OpenBCIHub`

Let me know if it's running for you. I think you may need to compile and run the hub from source, just like you did with the GUI

JayroMartinez, 2019-04-18:
I'm not able to run the executable. So, how can I run the code from source? Because for the GUI is just to put it in the folder and open it with Processing. Should I do the same with the Hub code?

daniellasry, 2019-04-19:
I want to emphasize that Raspberry Pi is **not a supported platform**. None of us have tested what I'm about to tell you. There may be HUB dependencies that don't work on ARM processors. We don't know what the performance of the GUI is like on the raspberry Pi (I'm guessing poor). You are entering uncharted territory. You will probably run into some roadblocks along the way.

1. [Download and install node.js](https://nodejs.org/en/download/)
2. [Download the HUB's source code .zip (or clone the repository)](https://github.com/OpenBCI/OpenBCI_Hub/releases/latest)
3. Open a terminal. Then:
4. `cd` to your HUB source code directory (for example: `cd ~/Desktop/OpenBCI_Hub`)
5. Run `npm install`
6. Run `npm start`. This launches the hub.

If you run into errors feel free to post here, but I can't promise I'll know the answer.

Best of luck!

JayroMartinez, 2019-04-23:
Now I'm able to run the OpenBCI Hub, but when I try to connect  to the device (I select the Serial/COM port & click on "get channel") I get this error:

[DEFAULT]: Attempting to open Serial/COM port: /dev/ttyUSB0
Radios_Config: connect_to_portName: attempting to open serial port: /dev/ttyUSB0
Radios_Config: connect_to_portName: port is open!
Error reading from Serial/COM port

Any idea on how to fix this?

daniellasry, 2019-04-26:
I don't know how to fix this, but here's how to get a more detailed error messages:

1. Open the file src/background.js
2. find the lines: `const debug = false;` and `const verbose = false;`
3. change them to const debug = true;` and `const verbose = true;`
4. re-launch the hub, launch the GUI
5. look at the hub's console output for any errors while connecting

retiutut, 2019-05-05:
It doesn't look like FTDI provides an driver for ARM. https://www.ftdichip.com/Drivers/VCP.htm

daniellasry, 2019-05-05:
Hmm it says over there that VCP drivers are included with the Ubuntu kernel. Maybe installing ubuntu on the raspverry pi would help. But I don't think it's straightforward to install Ubuntu on a Pi

retiutut, 2019-05-05:
I think it uses Ubuntu-mate?

closed 2019-05-05 by retiutut