OpenBCI/OpenBCI_GUI #505: Keyboard Shortcuts don't work

Otemo, 2019-05-01:
## Problem
Keyboard Shortcuts don't work.
The "spacebar" seems to work fine but does not react to the other keys.

## Expected 
ex. I can activate or deactivate channels using key board
https://docs.openbci.com/OpenBCI%20Software/01-OpenBCI_GUI

## Operating System and Version
Windows 10

## GUI Version
v4.1.2-beta.1
OpenBCIHub v2.0.8

## Running standalone app
Rrunning from Processing 3

## Type of OpenBCI Board
Both Cyton and Ganglion

## Are you using a WiFi Shield?
No

retiutut, 2019-05-01:
@Otemo Thanks for making this issue so we can discuss this recent change in the GUI!

We are working to make the GUI more accessible for all users. To prevent some new users from accidentally changing board settings, we have added an "Expert Mode" toggle button in the "Settings" dropdown.

[image]

@Otemo, to fix your issue, set "Expert Mode On". This will be remembered in the GUI settings for each Data Mode.

Let me go ahead and update the Docs to reflect this change.

Otemo, 2019-05-01:
I changed the settings according to your instructions. Then the keyboard shortcuts work!
I have not been aware of the nice feature of the new GUI.

Thank you for your easy-to-understand explanation.

retiutut, 2019-05-03:
The docs have been updated! Closing this issue.

https://github.com/OpenBCI/Docs/blob/master/OpenBCI%20Software/01-OpenBCI_GUI.md

closed 2019-05-03 by retiutut