OpenBCI/OpenBCI_GUI #506: Unstable Digital read on version 4.1.2(beta)

lopezvaa, 2019-05-03:
## Problem

I am attempting to set up an external trigger through my cyton board. However, when switching on the digital read via the GUI, I am getting seemingly random numbers from D11, 12, 13, 17,18. (D18 hovers around 8000 and seems to be an accelerometer).  Pushing the prog button should turn D17 from 0 to 1; however, it presents random numbers.

[image]

The Raw output files does not record the digital channesl instead, it is as if the accelerometer data was being sampled.

[image]

## Expected 

I expected to see the Digital reads as 0s and 1s and see a state change with a TTL level change (or pressing the prog button) and the raw files to contain the 5 columns for the digital inputs rather than the three accelerometer ones.

## Operating System and Version
Windows 10 Enterprise 64 bi

## GUI Version

4.1.2 (beta)

## Running standalone app

Standalone app

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

lopezvaa, 2019-05-03:
I ran the GUI code in Processing. There, the digital read functionallity works fine and the raw file outputs the digital channels as it is meant to. So, the issue is narrowed down to the stand alone release v4.1.2 (beta) that I downloaded while trying to get my wifishield to work,

Rolling back to release 4.0.4 fixed it (the latest stable release from the Download section is 4.0.03 also works)

Kind regards
Alejandro

retiutut, 2019-05-03:
@lopezvaa Thanks for the screenshots, looking into this soon.

retiutut, 2019-05-05:
I don't think the GUI code has changed. Possibly Hub related.

retiutut, 2019-05-05:
At least the older version works, so there is a standard of what should be happening. I can see that the current values in 4.1.2 are indeed accelerometer data instead of digital read.

retiutut, 2019-05-05:
I might have found the problem. Maybe digital read is turned on in the GUI but the wrong code is sent to the board.

retiutut, 2019-05-05:
Fixed 🔥 
[image]

lopezvaa, 2019-05-07:
Great @retiutut ! thanks for the updates and the prompt response. Should I close the Issue or do you do that yourself?

retiutut, 2019-05-07:
I'll close the issue when the Pull Request is done.

PR #508 (2019-05-09): Dev 4.1.2 Minor UI Fixes and Edits
- [x] help dropdown needs to be shifted over

- [x] UI: Fix Help Button and Settings Button Overlap using Button.ignoreHover

- [x] Fix #418
- [x] Fix #493, but also delete invalid files when detected. Currently, it will be remade when user closes or stops GUI, or saves settings.

- [x] Fix #506 and #422 by setting BoardMode using integers

- [x] Fix #509 

____

- [x] update changelog et al and bump version to 4.1.2-beta.2
commit 689c0bb3 (2019-05-09): Fix Bug: Send board mode as integer for Cyton Fixes #506
commit 8401ccf2 (2019-05-09): Fix Bug: Send board mode as integer for Cyton Fixes #506
closed 2019-05-08 by retiutut