OpenBCI/OpenBCI_GUI #509: Match Time Series drop-down behavior to other graphs

sublimatedpotato, 2019-05-05:
## Problem

The drop-down menus in Time Series widget do not close or allow alternate selection after making a selection. The drop-down must be closed and re-opened to make an additional selection.

[image]

## Expected 

All other drop-downs for the other widgets appear to auto-close after a selection is made, so this is the expected behavior in Time Series, too.

## Operating System and Version

macOS

## GUI Version

4.1.2-beta.0

## Running standalone app

No

## Type of OpenBCI Board

None

## Are you using a WiFi Shield?

No

retiutut, 2019-05-08:
Thanks for catching this! Enforcing the style and functionality already in place!

closed 2019-05-08 by retiutut

sublimatedpotato, 2019-05-09:
No problem @retiutut . Thanks for getting it resolved! Had just opened the app for the first time to explore #315 and #180 and noticed this. Figured it could be an easy solution. Hoping to get more involved on the repo in general, though.

PR #508 (2019-05-09): Dev 4.1.2 Minor UI Fixes and Edits
- [x] help dropdown needs to be shifted over

- [x] UI: Fix Help Button and Settings Button Overlap using Button.ignoreHover

- [x] Fix #418
- [x] Fix #493, but also delete invalid files when detected. Currently, it will be remade when user closes or stops GUI, or saves settings.

- [x] Fix #506 and #422 by setting BoardMode using integers

- [x] Fix #509 

____

- [x] update changelog et al and bump version to 4.1.2-beta.2