OpenBCI/OpenBCI_GUI #510: Playback not showing time series

jps2000, 2019-05-06:
## Problem

Playback shows accelerometer, time, bar  etc but no time series plots are coming

## Expected 

should work like older versions did

Windows

macOS/Windows/Linux

## GUI Version

V4.1.2.beta1

## Running standalone app

Are you running the downloaded app or are you running from Processing 3

## Type of OpenBCI Board

Ganglion

## Are you using a WiFi Shield?

No

retiutut, 2019-05-06:
@jps2000 Can you provide a screenshot or more details? Thanks.

jps2000, 2019-05-07:
Tomorrow
just make a small recording with ganglion. The txt file contains voltages in channels and correct file format.
Time series however , show nicely playback slider, running time etc but flatline channels.
the same holds for old recordings.

retiutut, 2019-05-07:
@jps2000 I'm not seeing this issue. 

I just made a short recording using Ganglion, and it opens in Playback Mode with no errors.

Any more details or screenshots?

jps2000, 2019-05-08:
@retiutut 
I re- tested it on my computer and could not reproduce the effect (like you did). 
It was noticed on a computer of my colleage.
We will figure out what may have caused the problem.
I come back as soon we know more.
thank you

jps2000, 2019-05-09:
I found that the software was not installed properly. 
Now it works. Sorry for inconveniences.

closed 2019-05-09 by jps2000