OpenBCI/OpenBCI_GUI #511: LSL streaming into BCILAB

poochi41, 2019-05-07:
## Problem

So I'm playing around with the new beta build of the GUI and the LSL network widget seems to be doing it's job until I actually stream real time data to the BCILAB LSL stream inlet.

For example, if I use the OpenBCI GUI to run simulated data and then stream via the LSL protocol  to the BCILAB program, it will render the simulated data quite well using the "vis_stream" command in MATLAB and assigning the appropriate values in the pop up window. Once the graphing window appears you can scale the sizing of the graph with the up and down arrow keys to match what you have in the time series window in the OpenBCI GUI.

However, if I physically start using the headset to record live readings and then send the data to BCILAB in MATLAB, the graphing does NOT look correct. I am wondering if that has something to do with the "metadata" or "labeling" of the data stream coming OUT of the OpenBCI GUI.

I'm just thinking out loud here, but maybe you guys could help me understand what's going on a little bit better. I would really appreciate it!

## Expected 

In a perfect world, what do you expect to happen.

## Operating System and Version

Windows 10 Home (Build 1809)

## GUI Version

Version v4.1.2-beta.0

## Running standalone app

I am running the standalone app!

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

retiutut, 2019-05-07:
I tested the LSL stream using OpenVibe on Windows 10.

retiutut, 2019-05-08:
I think I may have tested this already using the Cyton. LSL was almost completely broken in standalone! Also, I see in the code in networking that the TimeSeries data is sent as unfiltered. I'll test this again tomorrow using the latest development.

poochi41, 2019-05-08:
> I think I may have tested this already using the Cyton. LSL was almost completely broken in standalone! Also, I see in the code in networking that the TimeSeries data is sent as unfiltered, that may be why it looks different outside of the GUI. I'll test this again tomorrow using the latest development.

Let me know how it goes. I saw the buffered data in BCILAB looking like raw ADC values which is starting to make more sense. Keep me posted!

retiutut, 2019-05-08:
Looks fine! I can see eye blinks in Fp1 and Fp2 in both the GUI and OpenVibe. To note: the GUI is sending out raw(or unfiltered) EEG, so there is addition signal processing happening in the [OpenBCI/OpenVibe Example](https://github.com/OpenBCI/Docs/blob/master/assets/files/).

closed 2019-05-08 by retiutut