OpenBCI/OpenBCI_GUI #514: LSL stream cannot be started.

redbaloon, 2019-05-14:
## Problem

In OpenBCI_GUI, the LSL stream cannot be started. Pressing the "start" button makes the application's GUI freeze.

## Expected 

When the button start is pressed, the LSL stream starts and the OBCI_GUI's GUI keeps running.

## Operating System and Version

Linux ubuntu 16.10, Windows 10
[image]

## GUI Version

410, 407

## Running standalone app

Are you running the downloaded app or are you running from Processing 3
I am not running anything special, and the application is not running from Processing 3.

## Type of OpenBCI Board

Ganglion

## Are you using a WiFi Shield?

No

## Log
`
java.lang.InterruptedException: sleep interrupted
LSL selected from Protocol Menu
Hub: apparent sampleIndex jump from Serial data: 16 to  21.  Keeping packet. (112)
numPacketsDropped = 5
Hub: apparent sampleIndex jump from Serial data: 156 to  161.  Keeping packet. (113)
numPacketsDropped = 5
Hub: apparent sampleIndex jump from Serial data: 6 to  11.  Keeping packet. (114)
numPacketsDropped = 5
Stream update numChan to 4
nPointsPerUpdate 8
dataToSend len: 32
[TimeSeries, obci_eeg1, EEG, 4, 1]
java.lang.RuntimeException: java.lang.NoClassDefFoundError: com/sun/jna/Platform
	at processing.opengl.PSurfaceJOGL$2.run(PSurfaceJOGL.java:412)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NoClassDefFoundError: com/sun/jna/Platform
	at edu.ucsd.sccn.LSL.<clinit>(LSL.java:976)
	at edu.ucsd.sccn.LSL$StreamInfo.<init>(LSL.java:123)
	at OpenBCI_GUI$Stream.openNetwork(OpenBCI_GUI.java:28160)
	at OpenBCI_GUI$Stream.start(OpenBCI_GUI.java:27571)
	at OpenBCI_GUI$W_networking.startNetwork(OpenBCI_GUI.java:27389)
	at OpenBCI_GUI$W_networking.mouseReleased(OpenBCI_GUI.java:27113)
	at OpenBCI_GUI$WidgetManager.mouseReleased(OpenBCI_GUI.java:28957)
	at OpenBCI_GUI.mouseReleased(OpenBCI_GUI.java:12447)
	at processing.core.PApplet.mouseReleased(PApplet.java:2777)
	at processing.core.PApplet.handleMouseEvent(PApplet.java:2686)
	at processing.core.PApplet.dequeueEvents(PApplet.java:2599)
	at processing.core.PApplet.handleDraw(PApplet.java:2440)
	at processing.opengl.PSurfaceJOGL$DrawListener.display(PSurfaceJOGL.java:866)
	at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
	at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
	at jogamp.opengl.GLAutoDrawableBase$2.run(GLAutoDrawableBase.java:443)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
	at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
	at com.jogamp.newt.opengl.GLWindow.display(GLWindow.java:759)
	at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:81)
	at com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:452)
	at com.jogamp.opengl.util.FPSAnimator$MainTask.run(FPSAnimator.java:178)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
Caused by: java.lang.ClassNotFoundException: com.sun.jna.Platform
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:349)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 24 more
Hub: apparent sampleIndex jump from Serial data: 168 to  173.  Keeping packet. (115)
numPacketsDropped = 5
Hub: apparent sampleIndex jump from Serial data: 74 to  77.  Keeping packet. (116)
numPacketsDropped = 3
`

wjcroft, 2019-05-14:
Please try running the GUI from Processing IDE, rather than using the standalone,

https://docs.openbci.com/OpenBCI%20Software/01-OpenBCI_GUI#the-openbci-gui-running-the-openbci-gui-from-the-processing-ide

Regards,

William

redbaloon, 2019-05-14:
Hi wjcroft!

 > Please try running the GUI from Processing IDE, rather than using the standalone,

Running the GUI from Processing is not an option, in case of workshops for example. Processing really complicate the setup. Also, I had run into other problems trying to make OBCI_GUI run from Processing IDE. Thanks anyway wjcroft.

In order to face one problem at time.. it seems the 412 fix the problem. I tested it on Linux 16.10 and:

`Networking: LSL selected from Protocol Menu
++++Opening dropdown dataType1
Stream update numChan to 4
nPointsPerUpdate 8
dataToSend len: 32
StringList size=5 [ "TimeSeries", "obci_eeg1", "EEG", "4", "1" ]
[DEFAULT]: Network Stream Started` 
..and the GUI doesn't freeze!

https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v4.1.2-beta.2
(fixed already in v4.1.2-beta.0)

closed 2019-05-14 by redbaloon

retiutut, 2019-05-14:
@redbaloon There was a missing dependency for the LSL feature, including the library in the standalone explicitly fixed the issue. Thanks for confirming this!