OpenBCI/OpenBCI_GUI #529: LSL doesn't send stream correctly

PMK89, 2019-05-26:
## Problem

Trying to receive the LSL stream rather with pylsl example or NFBlab results in:

TypeError: 'list' object is not callable

It seems similar to issue #148

## Expected 

Receiving a TimeSeries LSL Stream

## Operating System and Version

Kubuntu 19.10 Kernel 5.0.0-15-generic

## GUI Version

4.1.2 and 4.1.2-beta.2

## Running standalone app

Standalone, Processing spawns errors

## Type of OpenBCI Board

Cyton (Problem also occurs with Synthetic data)

## Are you using a WiFi Shield?

No

## Console Log

>>> from pylsl import StreamInlet, resolve_stream
>>> streams = resolve_stream('type', 'EEG')
>>> inlet = StreamInlet(streams[0])
Note: The stream named '(invalid: bad lexical cast: source type value could not be interpreted as target)' could not be recovered automatically if its provider crashed because it does not specify a unique data source ID.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pmk/anaconda3/envs/nfb/lib/python3.5/site-packages/pylsl/pylsl.py", line 677, in __init__
    self.sample = self.sample_type()
TypeError: 'list' object is not callable

retiutut, 2019-05-28:
That does not look like a console log from the GUI. I'm not sure this is a problem with the GUI, since LSL streaming works with OpenVibe just fine.

retiutut, 2019-05-28:
```
    } else if (this.protocol.equals("LSL")){

                float[] avgPowerLSL = new float[numChan*numBandPower];
                for (int i=0; i<numChan;i++){
                    for (int j=0;j<numBandPower;j++){
                        dataToSend[j+numChan*i] = dataProcessing.avgPowerInBins[i][j];
                    }
                }
                outlet_data.push_chunk(dataToSend);
```

The above code correctly streams unfiltered EEG data. Leaving this issue closed unless more information is presented.

retiutut, 2019-05-28:
Re-opening this issue until testing has been completed.

miladinovic, 2019-07-02:
So, on Ubuntu still does not work, even with OpenVIBE acquisition server. I downloaded the version v4.1.3-beta2 from the development branch and run it in the processing on Ubuntu 18.04.

In the processing seems everything fine:

`[DEFAULT]: Data stream started.
New widget [3] selected for container...
Your cursor is in Accelerometer
Networking: LSL selected from Protocol Menu
Stream update numChan to 16
nPointsPerUpdate 10
dataToSend len: 160
StringList size=5 [ "TimeSeries", "EEG", "EEG", "16", "0" ]
[DEFAULT]: Network Stream Started`

But the data cannot be detected in OpenVIBE acquisition server (no stream id)
http://i66.tinypic.com/332ttgx.png

The same issue happens with the pylsl library

`user~$ python
Python 2.7.15+ (default, Nov 27 2018, 23:36:35) 
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from pylsl import StreamInlet, resolve_stream
>>> streams = resolve_stream('type', 'EEG')
>>> inlet = StreamInlet(streams[0])
Note: The stream named '(invalid: bad lexical cast: source type value could not be interpreted as target)' could not be recovered automatically if its provider crashed because it does not specify a unique data source ID.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/bioingts/.local/lib/python2.7/site-packages/pylsl/pylsl.py", line 677, in __init__
    self.sample = self.sample_type()
TypeError: 'list' object is not callable
>>> 
`

retiutut, 2019-07-02:
@miladinovic Thanks for the info and awesome screenshot! It looks like you are using OpenBCI GUI and OpenVibe correctly. Also, thanks for the additional info.

Reopening this issue and plan to test using Linux and OpenVibe.

miladinovic, 2019-07-03:
So, it seems that the problem is missing **liblsl64.so** (maybe consider including with the software). After installing https://github.com/sccn/liblsl/releases (**liblsl-1.13.0-Linux64-bionic.deb** for Ubuntu 18.04) and copying files from `/usr/LSL/lib/liblsl64.so` and `/usr/LSL/lib/liblsl64.so.1.13.0` to `/usr/lib/` (or making symbolic link if you prefer), everything works like a charm!

retiutut, 2019-07-03:
I see they are using MIT license, so it is possible to bundle this with the OpenBCI GUI!

retiutut, 2019-07-03:
@miladinovic I'm having severe issues building OpenVibe....... This is why I've only been testing this on Windows.

retiutut, 2019-07-03:
The title of this issue is "LSL doesn't send stream correctly". @miladinovic If you claim this is fixed by adding files in another directory outside of the OpenBCI GUI, there is no issue with the OpenBCI GUI.

closed 2019-07-03 by retiutut