OpenBCI/OpenBCI_GUI #531: Hub doesn't start on first try

andreaortuno, 2019-05-29:
## Problem

OpenBCI Hub doesn't initialize correctly when I first open the GUI. It works if I restart the GUI.

## Expected 

The OpenBCI Hub starts on the first try.

## Operating System and Version

Windows 10 64bit

## GUI Version

Latest release. 

## Running standalone app

Downloaded App

## Type of OpenBCI Board

N/A

## Are you using a WiFi Shield?

No

## Console Log

```
Screen Resolution: 1280 X 720
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://docs.openbci.com/OpenBCI%20Software/
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Windows
Local Version: 412, Latest Version: 412
GUI is up to date!
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
The file "SavedData/Settings/UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
java.net.ConnectException: Connection refused: connect
	at java.net.DualStackPlainSocketImpl.connect0(Native Method)
	at java.net.DualStackPlainSocketImpl.socketConnect(DualStackPlainSocketImpl.java:79)
	at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:350)
	at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
	at java.net.PlainSocketImpl.connect(PlainSocketImpl.java:172)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at java.net.Socket.connect(Socket.java:538)
	at java.net.Socket.<init>(Socket.java:434)
	at java.net.Socket.<init>(Socket.java:211)
	at processing.net.Client.<init>(Unknown Source)
	at OpenBCI_GUI$Hub.startTCPClient(OpenBCI_GUI.java:10598)
	at OpenBCI_GUI$Hub.<init>(OpenBCI_GUI.java:10574)
	at OpenBCI_GUI.systemUpdate(OpenBCI_GUI.java:1099)
	at OpenBCI_GUI.draw(OpenBCI_GUI.java:532)
	at processing.core.PApplet.handleDraw(PApplet.java:2475)
	at processing.opengl.PSurfaceJOGL$DrawListener.display(PSurfaceJOGL.java:866)
	at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
	at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
	at jogamp.opengl.GLAutoDrawableBase$2.run(GLAutoDrawableBase.java:443)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
	at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
	at com.jogamp.newt.opengl.GLWindow.display(GLWindow.java:759)
	at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:81)
	at com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:452)
	at com.jogamp.opengl.util.FPSAnimator$MainTask.run(FPSAnimator.java:178)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
[WARN]: Failed to connect to OpenBCIHub background application. LIVE functionality will be disabled.
Instantiating hub object...
Copying console log to clipboard!

```

retiutut, 2019-06-19:
This is hard to reproduce, so it may take longer to develop a fix.

retiutut, 2019-08-12:
Solution: Allow up to 14 seconds with a try every 2 seconds. Plenty of time to launch the hub directly from VSCode.

PR #571 (2019-08-12): 4.1.4-beta.0 UI edits
- Fixes #546
- Fixes #531
commit 3afbb71b (2019-08-12): Allow 14secs to connect to Hub  w/ try evry 2secs

Fixes  #531
closed 2019-08-12 by retiutut