OpenBCI/OpenBCI_GUI #546: Playback does not increment seconds played on file anymore

andrewjaykeller, 2019-06-20:
## Problem

The number of seconds is not incrementing during playback. 

See gif. The time stays still at 0 of 439. 

[image]

## Expected 

I want to know how long into a file I've been viewing. 

## Operating System and Version

macOS

## GUI Version

4.1.2

## Running standalone app

Standalone

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

retiutut, 2019-06-20:
@andrewjaykeller I have a file of similar length. I notice that the timestamp that shows is different than the gif you provided. 🤔
[image]

Here is one line of data from the OpenBCI txt file that I used:
`1, -174642.64, -178621.84, -180253.73, -180204.39, -175987.48, -176708.77, -177009.58, -176358.23, -174020.59, -180081.20, -174127.22, -176156.64, -176351.25, -180315.05, -173468.75, -174344.83, 0.000, 0.000, 0.000, 16:54:58.948, 1554674098948`

Can you share one line of data from the file that produced this error?

andrewjaykeller, 2019-06-21:
Hey!! 

yea so i'll do you one better by giving you two lines!

```
%Raw EEG Data
%Number of channels = 8
%Sample Rate = 250.0 Hz
%First Column = SampleIndex
%Last Column = Timestamp 
%Other Columns = EEG data in microvolts followed by Accel Data (in G) interleaved with Aux Data
0, 559.12887616, 561.565216256, 499.69558886399994, 478.57319078399996, 534.9666408960001, 578.306672512, 483.3117605119999, 536.2406903039999, 0, 0, 0, 523663, 1561048343962
1, 559.866483712, 562.191065088, 500.76847257599997, 481.36715878399997, 535.8383589119999, 578.641948672, 483.423519232, 535.7489519359999, 0, 0, 0, 523667, 1561048343966
2, 558.257158144, 561.006422656, 499.78499583999997, 481.188344832, 536.4418559999999, 579.4913149439999, 484.02701632, 536.10657984, 0, 0, 0, 523671, 1561048343970
```

andrewjaykeller, 2019-06-21:
Ugh this is probs my fault for changing the pretty date with board time... sssss

retiutut, 2019-06-21:
@andrewjaykeller When was that file made? I see more than one difference between your file and my (latest GUI) file.

- More than two decimal points for the EEG data
- A fourth column with some value I don't understand after Accel data `0, 0, 0, 523663,`
- Formatted timestamp

__
Similarity
- Header format is still the same
- Each line ends with unformatted timestamp

retiutut, 2019-06-22:
Details on the OpenBCI Data Format Update:
https://openbci.com/index.php/forum/#/discussion/2189/gui-saveddata-is-moving

retiutut, 2019-07-29:
@andrewjaykeller 

```
%Raw EEG Data
%Number of channels = 8
%Sample Rate = 250.0 Hz
%First Column = SampleIndex
%Last Column = Timestamp 
%Other Columns = EEG data in microvolts followed by Accel Data (in G) interleaved with Aux Data
0, 559.12887616, 561.565216256, 499.69558886399994, 478.57319078399996, 534.9666408960001, 578.306672512, 483.3117605119999, 536.2406903039999, 0, 0, 0, 523663, 1561048343962
1, 559.866483712, 562.191065088, 500.76847257599997, 481.36715878399997, 535.8383589119999, 578.641948672, 483.423519232, 535.7489519359999, 0, 0, 0, 523667, 1561048343966
2, 558.257158144, 561.006422656, 499.78499583999997, 481.188344832, 536.4418559999999, 579.4913149439999, 484.02701632, 536.10657984, 0, 0, 0, 523671, 1561048343970
```

> Does the timestamp come from the last column or the second to last column?

What do you think the second to last column is? Milliseconds since the GUI started?

retiutut, 2019-08-12:
Just fixed this!

PR #571 (2019-08-12): 4.1.4-beta.0 UI edits
- Fixes #546
- Fixes #531
closed 2019-08-12 by retiutut