OpenBCI/OpenBCI_GUI #553: Unable to establish link to Hub

DanielHHowell, 2019-07-03:
## Problem

Receiving the error "Unable to establish link to Hub. Live functionality will be disabled". 

I've determined the issue to be with the hub itself, when I try to run it I receive the error "/lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found", however the most up to date version of libc for Ubuntu 18.04 LTS is 2.27, and upgrading past that could break things if I understand correctly. So why is the Hub requiring a version of libc not supported on the current Ubuntu LTS? Thanks in advance!

## Expected 

Hub connection!

## Operating System and Version

Linux (Ubuntu 18)

## GUI Version

4.1.2.

## Running standalone app

Processing

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

## Console Log

[ERROR] "Unable to establish link to Hub. Live functionality will be disabled"

retiutut, 2019-07-03:
Sorry, but you've contradicted yourself.

> I've determined the issue to be with the hub itself

> I am not confident this is actually an issue with the OpenBCIHub as it is my system but not sure where else to turn

I can try to look into this...

retiutut, 2019-07-03:
"Unable to establish link to Hub. Live functionality will be disabled"  -- appears when the Hub takes too long to load or fails to launch

retiutut, 2019-07-03:
Not able to replicate this issue... probably an issue with your system.

closed 2019-07-03 by retiutut

retiutut, 2019-07-03:
If you really want to, you could try running the OpenBCIHub directly from Visual Studio Code. Just tested Hub 2.0.8 (again). No errors.

retiutut, 2019-07-03:
https://askubuntu.com/questions/421642/libc-so-6-version-glibc-2-14-not-found

retiutut, 2019-07-03:
Furthermore, searching the Hub code shows that "glib" is not mentioned in code or dependencies.

retiutut, 2019-07-03:
https://www.linode.com/docs/security/security-patches/patching-glibc-for-the-ghost-vulnerability/

DanielHHowell, 2019-07-04:
I appreciate all the help! Yes the standalone works fine, it's only when launching the hub in attempt to run the processing IDE. Have now tried this on 3 separate Ubuntu machines and receiving the same error - libc6 is up to date at 2.27 but it is requesting 2.28. Thanks for the assistance in any case

DanielHHowell, 2019-07-04:
"Having downloaded and extracted the binary release, one of the modules it uses indeed seems to have a GLIBC_2.28 symbol:

$ mkdir app
$ asar e resources/app.asar app
$ grep -r GLIBC_2.28
Binary file app/node_modules/serialport/build/Release/obj.target/serialport.node matches
Binary file app/node_modules/serialport/build/Release/serialport.node matches
Binary file app/node_modules/@serialport/bindings/build/Release/obj.target/bindings.node matches
Binary file app/node_modules/@serialport/bindings/build/Release/bindings.node matches
Binary file resources/app.asar matches"

retiutut, 2019-07-04:
That's cool, but you are talking about another package, and in the binaries.

retiutut, 2019-07-04:
I would recommend finding a way to make sure the Hub can launch directly from Visual Studio Code.

DanielHHowell, 2019-07-04:
Got it - will give that a shot, thank you.

retiutut, 2019-07-11:
@DanielHHowell I have reproduced the error. It's a problem with the 2.0.9 build that is currently posted. When I built the Hub again, it works fine.

To fix this, we are going to upload a new 2.0.9 version of the hub with no changes.

Thank you for being patient. Take care!

retiutut, 2019-07-11:
The following error occurs using the current, and soon to be replaced, 2.0.9 Hub for Linux only:
[image]

retiutut, 2019-07-11:
https://github.com/OpenBCI/OpenBCI_Hub/issues/99

retiutut, 2019-07-11:
Getting fixed today or tomorrow. Also, release 4.1.3-beta.2 is happening today/tomorrow.

retiutut, 2019-07-11:
This should work for you!

https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v4.1.3-beta.2