OpenBCI/OpenBCI_GUI #554: Open BCI abnormally high amplitude very often

rakansha, 2019-07-03:
## Problem

I have been using Open BCI Ganglion for quite a long time now(7 months). I purchased a new board and started using it a month ago. I am using Gold cup electrodes. Since a week it has started giving more amplifier saturations and sometimes it samples very slow. Typically, now the data can look like:
[image]

in case of slowing down.
In case of amplifier saturations that I am talking about, the high amplitude double spike on left:
[image]

It would have been acceptable if it happened rarely, but now it turns up every minute or so and I am using it for an important clinical study. This is affecting my project now. any ideas what is the problem here? I am using Lab Streaming Layer to stream it and reading it in Open Vibe
## Expected 

In a perfect world, what do you expect to happen.

## Operating System and Version

Windows 10

GUI Version latest 4.1.2 Beta

The version is displayed on startup.

Running standalone app

## Type of OpenBCI Board

Ganglion

## Are you using a WiFi Shield?

No

## Console Log

Paste any relevant text from the console window here

retiutut, 2019-07-03:
This has nothing to do with the OpenBCI GUI. There is no bug or issue here.

retiutut, 2019-07-03:
If this Ganglion is a legitimate OpenBCI board, you can contact support.

closed 2019-07-03 by retiutut

retiutut, 2019-07-03:
This is probably caused by connection issues. The Ganglion boards are extremely reliable and resilient.

retiutut, 2019-07-03:
I've recently covered a ganglion in aluminum foil and walked around the house. There are no spikes or abnormally high amplitudes if no electrodes are connected.

jps2000, 2019-07-04:
I also found similar spikes long time ago and want to share my experiences with it.
I 3d print a box for the ganglion in ABS. It works fine even without electrodes connected. then I noticed when touch or rub the box with  hands or garmet (wool, hair!) then I got such spikes / saturation.
This was caused by electrical charge of the ABS. 
Solution was to paint the inside of the box with conductive lacquer and connect it to ground. I leave  the part of the box close to  the Simblee module free  of course.

I used this copper spray:  
https://www.distrelec.de/de/leitfaehiger-beschichtungs-200-ml-kontakt-chemie-emi-35-200-ml-ml/p/30025848
I did not spray but  paint it with a small brush instead. (shake well, spray it into a small cup and then paint richly).
When dry you achieve <100 Ohms or better.