OpenBCI/OpenBCI_GUI #555: testing 4.1.2 with wi-fi shield

krisztian-hofstadter-tedor, 2019-07-05:
Hello, 

This is the first time I use the wi-fi shield with the open_bci gui. I need to start the system 3 times, before it start working. Video maybe useful: 

https://youtu.be/umOnsvqKQhk

Mac OS Sierra 10.12.6
Macbook Pro 2015
OpenBCI_GUI 10.12.6
Cyton firmware: 3.1.1
Wi-fi firmware: 2.0.5

Cheers! k

retiutut, 2019-07-09:
@krisztian-hofstadter-tedor I think you see now why we put the warning on the WiFi Shield. This is likely an error happening in the Hub.

https://github.com/OpenBCI/Docs/blob/master/Hardware/11-Wifi.md#about-the-wifi-shield

Try UDPx3 protocol. Also try turning the Bandpass Filter Off.

I use the Ganglion+WiFi shield only, so thanks for testing the Cyton+WiFi and documenting the results! 🤠

krisztian-hofstadter-tedor, 2019-07-09:
cheers, I'll test again asap.

krisztian-hofstadter-tedor, 2019-07-10:
hi! 

I only managed to test the serial communication, where I need to restart the system the same way, but only once instead of 3 times (with the wi-fi). I will test the wifi asap. 

here a video 
https://youtu.be/OG-AcfYX0cA

// maybe useful; 

and two logs: 

open_bci_log_01.rtf

```
Screen Resolution: 1920 X 1200
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://docs.openbci.com/OpenBCI%20Software/
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Mac
java.net.UnknownHostException: github.com
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:184)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at sun.security.ssl.SSLSocketImpl.connect(SSLSocketImpl.java:666)
	at sun.security.ssl.BaseSSLSocketImpl.connect(BaseSSLSocketImpl.java:173)
	at sun.net.NetworkClient.doConnect(NetworkClient.java:180)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:463)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:558)
	at sun.net.www.protocol.https.HttpsClient.<init>(HttpsClient.java:264)
	at sun.net.www.protocol.https.HttpsClient.New(HttpsClient.java:367)
	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.getNewHttpClient(AbstractDelegateHttpsURLConnection.java:191)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1156)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:1050)
	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:177)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1564)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1492)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:347)
	at processing.core.PApplet.createInputRaw(PApplet.java:7112)
	at processing.core.PApplet.createInput(PApplet.java:7065)
	at processing.core.PApplet.loadStrings(PApplet.java:7519)
	at OpenBCI_GUI$TopNav.loadGUIVersionData(OpenBCI_GUI.java:14695)
	at OpenBCI_GUI$TopNav.<init>(OpenBCI_GUI.java:14341)
	at OpenBCI_GUI.delayedSetup(OpenBCI_GUI.java:454)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at processing.core.PApplet.method(PApplet.java:3861)
	at processing.core.PApplet$2.run(PApplet.java:3899)
The file "https://github.com/OpenBCI/OpenBCI_GUI/releases/latest" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
The file "/Users/khofstadter/Documents/OpenBCI_GUI/Settings/UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
Instantiating hub object...
Hub: clientEvent: handshake complete
"SansSerif.plain" is not available, so another font will be used. Use PFont.list() to show available fonts.
"SansSerif.bold" is not available, so another font will be used. Use PFont.list() to show available fonts.
Refreshing port list...
sourceList: click! 
setInterface: curInterface: -1
channel count set to 8
Setting Latency to 10000
Setting WiFi Internet Protocol to tcp
Setting WiFi style to dynamic
[DEFAULT]: Protocol Serial Selected for Cyton
setInterface: curInterface: 0
Setting sample rate for Cyton to 250Hz
[... 231 lines cut ...]
OpenBCI_GUI: deactivating channel 5
Hub: sendCommand(char): sending '5'
Hub: processCommand: success -- 130634
[6] onOff pressed
[6] onOff released
OpenBCI_GUI: deactivating channel 6
Hub: sendCommand(char): sending '6'
Hub: processCommand: success -- 131861
[7] onOff pressed
[7] onOff released
OpenBCI_GUI: deactivating channel 7
Hub: sendCommand(char): sending '7'
Hub: processCommand: success -- 132418
[8] onOff pressed
[8] onOff released
OpenBCI_GUI: deactivating channel 8
Hub: sendCommand(char): sending '8'
Hub: processCommand: success -- 133081
Hub: apparent sampleIndex jump from Serial data: 191 to  250.  Keeping packet. (1)
numPacketsDropped = 59
Hub: apparent sampleIndex jump from Serial data: 250 to  193.  Keeping packet. (2)
numPacketsDropped = 198
Copying console log to clipboard!

```1

retiutut, 2019-07-11:
Wow there is more than one error in the console. We are about to release 4.1.3. Give me some time to be at the computer and go through this. Thanks!

krisztian-hofstadter-tedor, 2019-07-11:
Hi! 

I did some tests with the wi-fi shield as well. 

As far as I understand the logbook can tell you the settings used, however here a rather long video: 

https://youtu.be/sym1AIkioUQ

I am a bit worried about the jumping signal around 50hz, do you know what that could be? 
// I'll ask on the forum as well;  

cheers! k

---

open_bci_log01-1

```
Screen Resolution: 1920 X 1200
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://docs.openbci.com/OpenBCI%20Software/
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Mac
java.net.UnknownHostException: github.com
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:184)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at sun.security.ssl.SSLSocketImpl.connect(SSLSocketImpl.java:666)
	at sun.security.ssl.BaseSSLSocketImpl.connect(BaseSSLSocketImpl.java:173)
	at sun.net.NetworkClient.doConnect(NetworkClient.java:180)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:463)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:558)
	at sun.net.www.protocol.https.HttpsClient.<init>(HttpsClient.java:264)
	at sun.net.www.protocol.https.HttpsClient.New(HttpsClient.java:367)
	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.getNewHttpClient(AbstractDelegateHttpsURLConnection.java:191)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1156)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:1050)
	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:177)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1564)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1492)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:347)
	at processing.core.PApplet.createInputRaw(PApplet.java:7112)
	at processing.core.PApplet.createInput(PApplet.java:7065)
	at processing.core.PApplet.loadStrings(PApplet.java:7519)
	at OpenBCI_GUI$TopNav.loadGUIVersionData(OpenBCI_GUI.java:14695)
	at OpenBCI_GUI$TopNav.<init>(OpenBCI_GUI.java:14341)
	at OpenBCI_GUI.delayedSetup(OpenBCI_GUI.java:454)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at processing.core.PApplet.method(PApplet.java:3861)
	at processing.core.PApplet$2.run(PApplet.java:3899)
The file "https://github.com/OpenBCI/OpenBCI_GUI/releases/latest" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
The file "/Users/khofstadter/Documents/OpenBCI_GUI/Settings/UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
Instantiating hub object...
Hub: clientEvent: handshake complete
"SansSerif.plain" is not available, so another font will be used. Use PFont.list() to show available fonts.
"SansSerif.bold" is not available, so another font will be used. Use PFont.list() to show available fonts.
sourceList: click! 
setInterface: curInterface: -1
channel count set to 8
Setting Latency to 10000
Setting WiFi Internet Protocol to tcp
Setting WiFi style to dynamic
[... 1092 lines cut ...]
[DEFAULT]: Wifi Device Name = OpenBCI-5125
wifiList: click! 
[DEFAULT]: Wifi Device Name = OpenBCI-5125
Calling initSystem()

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

OpenBCI_GUI: closing log file
OpenBCI_GUI: openNewLogFile: opened ODF output file: /Users/khofstadter/Documents/OpenBCI_GUI/OpenBCI-RAW-2019-07-11_14-54-53.txt
init analog read bar 0
init analog read bar 1
Hub: processConnect: made it -- 35277 code: 435
Error in processConnect: RESP_ERROR_WIFI_NEEDS_UPDATE
Hub: killAndShowMsg: WiFi Shield Firmware is out of date. Learn to update: docs.openbci.com/Hardware/12-Wifi_Programming_Tutorial
openBCI_GUI: haltSystem: Halting system for reconfiguration of settings...
Closing any open SD file. Writing 'j' to OpenBCI.
Failed to connect to data source... 3
[ERROR]: Failed to connect to data source. Check that the device is powered on and in range. Also, try pressing AUTOSCAN.
The file "/Users/khofstadter/Documents/OpenBCI_GUI/Settings/UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
Copying console log to clipboard!

```

krisztian-hofstadter-tedor, 2019-07-16:
further checks with wifi+cyton:

https://youtu.be/UiG-YJ7Kn4o

same system. 

issues: 
1. UDPx3 not connecting;
2. signal is still jumping around 50hz (UDP) even with filters off; 

thanks! k

retiutut, 2019-07-27:
@krisztian-hofstadter-tedor Try updating the Cyton firmware.

retiutut, 2019-07-27:
@krisztian-hofstadter-tedor https://github.com/OpenBCI/OpenBCI_GUI/issues/231 is now marked as "in progress"

retiutut, 2019-08-29:
@krisztian-hofstadter-tedor Not sure this will fix the WiFi shield issues for now, but here is the latest pre-release of the GUI:
https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v4.1.5-beta.3

Looking to fix some of the errors in the Logs you've shared for the next release.

krisztian-hofstadter-tedor, 2019-09-04:
thanks, I will check asap. k

bmartin427, 2019-09-11:
I am also not getting UDPx3 mode working, with Cyton + WiFi shield.  I am using GUI 4.1.5, Cyton FW 3.1.2, and WiFi FW 2.0.5.  As above, the GUI prints RESP_ERROR_WIFI_NEEDS_UPDATE, despite all versions being the newest.

Digging into the source, and snooping with tcpdump, it appears this error is internally generated by the hub when the shield itself reports a 404 error in response to a POST to /udpBurst (this behavior was introduced with hub version 1.3.7).  Digging further, it appears the URL should *not* be /udpBurst at all, but rather /udp, just with the 'redundancy' field set to true.  This seems to have been a bug that was fixed in hub version 1.3.9, but has now been reintroduced.  The original fix in that version of the hub seems to be to catch any requests from the GUI for 'udpBurst', and rewrite them to 'udp' instead.  At some point, the behavior was changed to just pass the GUI request through.  Should the GUI simply be requesting the right mode, rather than working around this in the hub?  Or should an issue be opened against the hub project to reintroduce this workaround?

bmartin427, 2019-09-11:
As a separate issue, I can also report that starting a regular UDP session is flaky, and only seems to work on the second try.  Additionally, some state in the GUI seems to get confused by starting the session twice, with the result that I cannot then stop the session.

bmartin427, 2019-09-11:
#594 addresses the UDP burst issue.  I'm looking into the connection flakiness now.

retiutut, 2019-09-11:
@bmartin427 Thanks looking into this also. 

> Should the GUI simply be requesting the right mode, rather than working around this in the hub? Or should an issue be opened against the hub project to reintroduce this workaround?

If you can fix any issues without modifying the Hub, that is quicker. If this requires modifying the Hub, then thats what we have to do. WiFi related issues are low-priority, so your help in this area is greatly appreciated.

bmartin427, 2019-09-11:
So initially, I can report that the flakiness is due to the 'j' command that the GUI issues at startup (disabling any previous SD logging).  Over serial, this command produces no output if SD logging were not already taking place.  The wifi shield responds to this lack of response from the cyton by holding the socket open for two seconds after the POST, and then either just silently closing the socket, or returning code 502 "timeout getting command response, be sure board is fully connected".  In response to this, the hub reports code 406, either forwarding the above string, or reporting "socket hang up" in the case of the silent disconnect.  Either one of these causes the GUI to report ERROR_COMMAND_NOT_ABLE_TO_BE_SENT, which aborts the session.

On a second attempt to start a session, it seems the GUI no longer attempts to send the 's' and 'j' commands, so startup succeeds.  However, clicking STOP SESSION no longer takes you back to the session settings dialog.

Right now, I'm digging through history, trying to figure out if this is recent breakage, and what changed to trigger it.  But generally speaking, the following fixes are possible:
 * The most robust solution would probably be to have the cyton firmware actually respond to the 'j' command.  However, I'm guessing that a firmware fix is probably the least desirable solution from a practicality standpoint.
 * The wifi shield firmware could report 200 OK for commands that are received over REST but trigger no cyton response.  This isn't really better than the first option and may in fact be a bit worse.
 * The GUI could ignore errors happening in response to the 'j' command exclusively.  This could be tricky, because the GUI/hub protocol doesn't currently provide any way to know *which* command a response code corresponds to, and the GUI sometimes has multiple commands outstanding simultaneously.
 * The GUI could ignore *all* command error codes.

bmartin427, 2019-09-11:
Alternatively, I have noticed that the serial command pathway never sends the 'j' command.  In that case, another solution would be just to not send this command if that is possible.

bmartin427, 2019-09-11:
Whoops.  Looks like the ERROR_COMMAND_NOT_ABLE_TO_BE_SENT was a red herring.  The 'j' command was only being sent during haltSystem(), which is called *after* something else has already gone wrong.  The line on the console that was meant to indicate this reads:
`null`
Which didn't clue me in at all that it had happened.  So, everything I said above is still problematic, but the real problem is that the firmware version being detected is 'null', which is causing an exception to be thrown from initSystem().

retiutut, 2019-09-11:
@bmartin427 I understand this back and forth during issue investigation all too well. 🙃

> Which didn't clue me in at all that it had happened. So, everything I said above is still problematic, but the real problem is that the firmware version being detected is 'null', which is causing an exception to be thrown from initSystem().

If you've traced this back to an uncaught exception in initSystem(), that sounds quite fixable. Take a few more looks at what's going on before we take action by changing GUI code.

bmartin427, 2019-09-11:
The line throwing the exception is line 823 in OpenBCI_GUI.pde:
`
    if (eegDataSource == DATASOURCE_CYTON && hub.getFirmwareVersion().equals("v1.0.0")) {
`
InterfaceHub's firmwareVersion got set to null during processConnect(), because evidently the key didn't exist in the json struct.  I'm still trying to figure out how that is *supposed* to work.

retiutut, 2019-09-11:
That's a harded check that didn't account for WiFi being null.

retiutut, 2019-09-11:
I don't really use WiFi shield on Cyton, but I guess I can try this out now.

retiutut, 2019-09-11:
[image]

retiutut, 2019-09-12:
> The line throwing the exception is line 823 in OpenBCI_GUI.pde:
> if (eegDataSource == DATASOURCE_CYTON && hub.getFirmwareVersion().equals("v1.0.0")) {
> InterfaceHub's firmwareVersion got set to null [...]

```
    if (eegDataSource == DATASOURCE_CYTON) {
        if (hub.getFirmwareVersion() == null && hub.getProtocol() == PROTOCOL_WIFI) {
            println("Cyton+WiFi: Unable to find board firmware version");
        } else if (hub.getFirmwareVersion() == "v1.0.0") {
            abandonInit = true;
        } else {
            //println("FOUND FIRMWARE FROM HUB == " + hub.getFirmwareVersion());
        }
    }
``` 
This allows it for now and will just print to console.

retiutut, 2019-09-12:
@bmartin427 Don't be surprised when you see https://github.com/OpenBCI/OpenBCI_GUI/issues/231, which is now officially the oldest open issue with the GUI, and caused me to reach out to OpenBCI in the first place. For this reason, I rarely plug up the Cyton+WiFi and I have fun using Ganglion+WiFi for EMG apps.

retiutut, 2019-09-20:
Using @bmartin427's fix in the Hub 2.1.0 (coming very soon).
[image]

bmartin427, 2019-09-20:
> > @bmartin427 Don't be surprised when you see #231, which is now officially the oldest open issue with the GUI, and caused me to reach out to OpenBCI in the first place. For this reason, I rarely plug up the Cyton+WiFi and I have fun using Ganglion+WiFi for EMG apps.

FWIW I've now taken a few minutes of EEG data with this setup, and didn't notice any periodic spikes.  I also don't hear any audible noise coming from my wifi shield the way some have reported.  Perhaps only some hardware is susceptible to this problem.

retiutut, 2019-11-13:
Also, if an internet connection is not available and you click "Update" the GUI will crash with a null pointer. Pushing code to fix this and prevent Exception from being thrown. 

This will go away
```
java.net.UnknownHostException: github.com
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:184)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at sun.security.ssl.SSLSocketImpl.connect(SSLSocketImpl.java:666)
	at sun.security.ssl.BaseSSLSocketImpl.connect(BaseSSLSocketImpl.java:173)
	at sun.net.NetworkClient.doConnect(NetworkClient.java:180)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:463)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:558)
	at sun.net.www.protocol.https.HttpsClient.<init>(HttpsClient.java:264)
	at sun.net.www.protocol.https.HttpsClient.New(HttpsClient.java:367)
	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.getNewHttpClient(AbstractDelegateHttpsURLConnection.java:191)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1156)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:1050)
	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:177)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1564)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1492)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:347)
	at processing.core.PApplet.createInputRaw(PApplet.java:7112)
	at processing.core.PApplet.createInput(PApplet.java:7065)
	at processing.core.PApplet.loadStrings(PApplet.java:7519)
	at OpenBCI_GUI$TopNav.loadGUIVersionData(OpenBCI_GUI.java:14695)
	at OpenBCI_GUI$TopNav.<init>(OpenBCI_GUI.java:14341)
	at OpenBCI_GUI.delayedSetup(OpenBCI_GUI.java:454)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at processing.core.PApplet.method(PApplet.java:3861)
	at processing.core.PApplet$2.run(PApplet.java:3899)
The file "https://github.com/OpenBCI/OpenBCI_GUI/releases/latest" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
```
and instead something like 

```
TopNav: Internet Connection Not Available
```

retiutut, 2019-11-13:
@krisztian-hofstadter-tedor Thank you for making this thread so that we were able to discuss the status of WiFi shield for GUI 4.1.2-4.1.6. We are preparing for GUI v5, so I think it's time to close this thread. Thanks for all the input everyone! 💯 🥇

closed 2019-11-13 by retiutut