OpenBCI/OpenBCI_GUI #557: App crashes by 16 chan bandpower send out through LSL (+solution)

Hzikajr, 2019-07-11:
## Problem & solution

I'm not sure how to make a pull request for now but have a solution for the problem I faced so post it as an issue here.
The GUI app crashes whenever I try to send band power data out through LSL. 8ch have no problem but 16ch has.
Using the latest developer branch, tested on macOS and Windows. Same error happened.

https://github.com/OpenBCI/OpenBCI_GUI/blob/11aab0d71110c3a77249dc9eed76879bfca2ec90/OpenBCI_GUI/W_Networking.pde#L1407
if you change the part `dataToSend[j+numChan*i]` to `dataToSend[j+numBandPower*i]` makes the code working.

Hope this helps.

retiutut, 2019-07-11:
That sounds like a typo!!!! Great find, easy fix. Looking into this and maybe we can fit this into 4.1.3(stable release)... In the process of building/releasing 4.1.3-beta.2

retiutut, 2019-07-11:
```
InitSettings: Saving Default Settings to file!
SoftwareSettings: 3 active widgets saved!
The file "/Users/Richard/Documents/OpenBCI_GUI/Settings/SynthSixteenUserSettings.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
InitSettings: /Users/Richard/Documents/OpenBCI_GUI/Settings/SynthSixteenUserSettings.json not found or other error.
[DEFAULT]: The new data source is SYNTHETIC (algorithmic) and NCHAN = [16]. User settings error: Default Settings Loaded.
New widget [3] selected for container...
Your cursor is in FFT Plot
Networking: LSL selected from Protocol Menu
[DEFAULT]: Data stream started.
Stream update numChan to 16
nPointsPerUpdate 10
dataToSend len: 160
StringList size=5 [ "BandPower", "obci_eeg1", "EEG", "16", "0" ]
[DEFAULT]: Network Stream Started
W_Networking.pde:1438:0:1438:0: ArrayIndexOutOfBoundsException: 160
```
**Replicated**

retiutut, 2019-07-11:
[image]

**Suggested Fix Works!**

retiutut, 2019-07-11:
Done 🔥🔥🔥

closed 2019-07-11 by retiutut

retiutut, 2019-07-11:
https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v4.1.3-beta.3

commit 0ab463ce (2019-07-11): update changelog for fix #557
commit a029c7e5 (2019-07-11): Merge pull request #559 from retiutut/hiccup-challenge

update changelog for fix #557