OpenBCI/OpenBCI_GUI #563: v4.1.3-beta.3 - not working: Unable to Load Jave Runtime Environment (Sierra OS)

krisztian-hofstadter-tedor, 2019-07-18:
## Problem

Explain the problem

## Expected 

In a perfect world, what do you expect to happen.

## Operating System and Version

macOS/Windows/Linux

## GUI Version

The version is displayed on startup.

## Running standalone app

Are you running the downloaded app or are you running from Processing?

## Type of OpenBCI Board

Cyton/Cyton+Daisy/Ganglion

## Are you using a WiFi Shield?

Yes/No

## Console Log

Paste any relevant text from the console window here

krisztian-hofstadter-tedor, 2019-07-18:
hi, 

was trying to test the newest release on Sierra but couldn't get it started:

'Unable to Load Java Runtime Environment' in pop up window.

more here if needed: https://youtu.be/T_WWrL1HDm0

thanks! k

retiutut, 2019-07-22:
Please just update your Mac. Supporting older versions of OS is a nightmare.

Mac OS Sierra Initial release: September 20, 2016; 2 years ago

retiutut, 2019-07-22:
@krisztian-hofstadter-tedor Please start opening the Console Log in these videos. It's very hard to understand what is happening without seeing the Console Log.

retiutut, 2019-07-22:
@krisztian-hofstadter-tedor As a consolation, I have seen the signal drop to zero sometimes as you have described in this video.

krisztian-hofstadter-tedor, 2019-07-22:
thanks Richard, 
- I'll try Direct mode; 
- update an OS - to Mojave; 
(keep the logs in mind for next time);

then, maybe my other issue with the 50hz jumps with wifi will be sorted as well; 
(would be good to find someone who works with cyton and wifi - this person could let me know firmware, os version, gui version, etc...) .. I think I am just worried that maybe one of the boards (cyton or wifi) is damaged. 

Thanks. k

retiutut, 2019-07-22:
@krisztian-hofstadter-tedor Let's focus on the simple java runtime error. Update your OS and let me know if the problem still happens.

There doesn't seem to be anything wrong with the boards.

This issue of dropped packets and artifacts is an extremely difficult issue: https://github.com/OpenBCI/OpenBCI_GUI/issues/231. In fact, this is what prompted me to start working on the GUI almost a year ago. If you want less dropped packets, use Cyton without the WiFi shield. I'm not interested in discussing this as it is "in progress".

krisztian-hofstadter-tedor, 2019-07-22:
aha. thanks for the clarification.

krisztian-hofstadter-tedor, 2019-07-23:
@retiutut

I have the same issue in a fresh Mojave install. Tried a few things, maybe this short video helps: 

https://youtu.be/eGPmWjdi73I

cheers! k

retiutut, 2019-07-23:
@krisztian-hofstadter-tedor Possibly caused by me building the app using an old version of Processing!

retiutut, 2019-07-23:
This issue has been replicated!

retiutut, 2019-07-23:
@krisztian-hofstadter-tedor Me and @leanneapichay have found a fix!!!!! 🔥🔥🔥

First Solution:
- Java 8 appears to have been updated to Version 8 Update 221, Release date July 16, 2019 https://www.java.com/en/download/
- Download and update Java, should allow the build you have already to run!

Second Solution:
- I/We build the OpenBCI_GUI.app and include the Java Runtime Environment, but this makes the App package much larger

krisztian-hofstadter-tedor, 2019-07-23:
@retiutut 

Cheers! I'll test 1st solution asap. k

retiutut, 2019-07-26:
@krisztian-hofstadter-tedor I've been told that 4.1.3-beta.3-BUILD-4-SIGNED works without having to reinstall java.

retiutut, 2019-07-26:
Closing this issue. @krisztian-hofstadter-tedor I'm about to bump the version to a stable release of 4.1.3, let me know if this happens again.

closed 2019-07-26 by retiutut

krisztian-hofstadter-tedor, 2019-07-29:
works without JAVA, thanks.