OpenBCI/OpenBCI_GUI #573: BandPower visualization widget does not work

Otemo, 2019-08-13:
## Problem
BandPower visualization widget could not be displayed in OpenBCI GUI.
When BandPower was choosed, GUI freezed.

## Expected 
BandPower can be choosed from a pull-down menu, then visualizaed.

## Operating System and Version
Windows10

## GUI Version
v4.1.4-beta.0
OpenBCIHub v2.0.9

## Running standalone app
running from Processing

## Type of OpenBCI Board
Ganglion

## Are you using a WiFi Shield?
No

[image]

retiutut, 2019-08-13:
@Otemo I have replicated this issue using Synthetic (4 Channels)! Should be able to fix this quickly. Thank you for testing the latest release!!!

retiutut, 2019-08-13:
[image]

retiutut, 2019-08-13:
This issue is being caused by the new ChannelSelect dropdown in the BandPower widget.

closed 2019-08-14 by retiutut

Otemo, 2019-08-14:
Hi retiutut,
Thank you for your quick support. I have confirmed that the GUI works properly!

PR #574 (2019-08-14): Make reusable ChannelSelect class in Widget.pde
- Closes #573 
- Consumed by BandPower, SSVEP, and any future widget in need of the feature
commit 3657536b (2019-08-14): Update changelog 4.1.4-beta.1 Fixes #573