OpenBCI/OpenBCI_GUI #577: Still getting RuntimeException

BeeCoughs, 2019-08-21:
## Problem

I am still getting the runtime exception when trying to launch the GUI from Processing. I have seen others mention this and supposedly it was fixed (#426 ), but it persists for me. I scrapped everything and followed the "RUNNING THE OPENBCI GUI FROM THE PROCESSING IDE" guide to a T multiple times just to be sure I set everything up right and I still am getting this error.

## Expected 
GUI Launches from Processing

## Operating System and Version
Windows 10

## GUI Version
4.1.3

## Running standalone app
From Processing

## Type of OpenBCI Board
Cyton

## Are you using a WiFi Shield?
No

## Console Log

java.lang.RuntimeException: Waited 5000ms for: <55bf7001, 5b7884c6>[count 3, qsz 1, owner <main-FPSAWTAnimator#00-Timer0>] - <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1>
	at processing.opengl.PSurfaceJOGL$2.run(PSurfaceJOGL.java:410)
	at java.lang.Thread.run(Thread.java:748)

daniellasry, 2019-08-22:
What hardware specs are you running on? This could narrow down whether it's a performance issue or something else. Does it freeze on a grey screen?

BeeCoughs, 2019-08-22:
I'm running on an Intel i7-8700K @3.7GHz with 64GB of ram and a GeForce 2080Ti (with up-to-date drivers). Plenty of HDD space (more than a TB) It freezes on a blank (white) window.

I also just double checked that the Electron Hub app is the suggested version (2.0.9) for this version of the GUI.

One bit in the readme that makes me wonder is the line that says "OpenGL acceleration is required". From what I can tell the NVidia driver I have supports OpenGL 4.6 but I suppose that doesn't necessarily mean I have acceleration enabled. Google is failing to give me a straight answer in terms of how I can figure out that info - is there some easy way I can check this?

Not sure if it's worth mentioning that the standalone application runs okay, with occasional freezing issues when the connection with the Cyton is interrupted.

Thanks for the help!

daniellasry, 2019-08-24:
Yeah if you have a 2080ti with nvidia drivers, you have acceleration enabled, no question about it. Even if you had no dedicated GPU the integrated graphics on the intel processor would give you GL acceleration.

Definitely good to know that the standalone app works. So there's something about running from processing...

Are you running the latest version of processing available (3.5.3)? I've had issues with older versions.

Could you paste the rest of the console log leading up to the text you pasted above? I want to see if there are any other failures that might be delaying startup.

daniellasry, 2019-08-29:
If you're in there modifying the code, you could pepper some print statements in the setup() function and figure out which part of setup is taking too long.

BeeCoughs, 2019-08-29:
Still having the same issue in VSCode unfortunately. I'm a little confused as to where exactly it's slowing down. The only part there's any appreciable time between lines printed to console is after "OpenBCI_GUI::Setup: Setup is complete!" and before the error.

I put print statements throughout the setup (and delayed_setup) functions and they seem to all run, but one I put right at the top of the draw function does not. I also put one in the top of the udpReceiveHandler function that does not get printed, though I can't tell if that would be expected or not. 

The console printout is exactly the same other than some of the numbers in the error:
```
java.lang.RuntimeException: Waited 5000ms for: <15aa3a48, 1b6dc68d>[count 3, qsz 1, owner <main-FPSAWTAnimator#00-Timer0>] - <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1>
        at processing.opengl.PSurfaceJOGL$2.run(PSurfaceJOGL.java:410)
        at java.lang.Thread.run(Thread.java:748)
```

I think I've thrown all I can at this. Thanks for taking the time to help regardless.

Joe-Westra, 2019-11-05:
I'm having the same issue as BeeCoughs, here are my specs:

**Operating System and Version**
Ubuntu 18.04.3

**GUI Version**
OpenBCI GUI 4.1.6
Running app via Processing IDE v 3.5.3

**Type of OpenBCI Board**
Cyton + Daisy

**Are you using a WiFi Shield?**
Yes

The RuntimeException is consistently thrown after GUI launch, selecting the cyton -> wifishield -> (setting parameters to UDP, 16 channels) -> pressing "Start Session".

Any ideas?

Hardware Specs
processor: AMD® Fx(tm)-8150 eight-core processor × 8 
memory: 16 G
HDD: 2TB

```
OpenBCI_GUI::Setup: Setup is complete!
Instantiating hub object...
hub IP:127.0.0.1
hub port:10996
Socket[addr=/127.0.0.1,port=10996,localport=35850]
Hub: CheckHubInit: The GUI is connected to the Hub!
sourceList: click! 
setInterface: curInterface: -1
channel count set to 8
Setting Latency to 10000
Setting WiFi Internet Protocol to tcp
Setting WiFi style to dynamic
[DEFAULT]: Protocol Wifi Selected for Cyton
setInterface: curInterface: 2
Setting sample rate for Cyton to 1000Hz

sample rate set to: 1000
[DEFAULT]: Transfer Protocol set to wifi
refreshWifiList
wifiList: click! 
[DEFAULT]: Wifi Device Name = OpenBCI-52EE
channel count set to 16
Setting WiFi Internet Protocol to udp
Settings: LogFileMaxDuration = 60 minutes
initButtonPressed: Calling initSystem()

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

OpenBCI_GUI: hub : Sent connect to Hub - Id: OpenBCI-52EE SampleRate: 1000Hz Latency: 10000ms
"Arial" is not available, so another font will be used. Use PFont.list() to show available fonts.
Hub: processConnect: made it -- 14970 code: 200
Hub: setBoardType(): sending 'daisy -- 14971
Hub: processBoardType: success -- 14972
[SUCCESS]: The GUI is done initializing. Settings Loaded! Press "Start Data Stream" to start streaming! --  Cyton firmware v3.1.1
InitSettings: Saving Default Settings to file!
SoftwareSettings: 3 active widgets saved!
LoadGUISettings: Applied Widget 0 to Container 4
LoadGUISettings: Applied Widget 2 to Container 9
Networking Data Types || show num dropdowns = 6
LoadGUISettings: Applied Widget 3 to Container 3
Hub: sendCommand(String): sending '/0'
Cyton: setBoardMode to :DEFAULT
LoadGUISettings: Expert Mode Off
Hub: processCommand: success -- 15588
Networking: Protocol mode set to UDP
Networking Data Types || show num dropdowns = 6
Apply UDP Networking Mode
Ch 0, Active, x24, Normal, Include, On, Off
Ch 1, Active, x24, Normal, Include, On, Off
Ch 2, Active, x24, Normal, Include, On, Off
Ch 3, Active, x24, Normal, Include, On, Off
Ch 4, Active, x24, Normal, Include, On, Off
Ch 5, Active, x24, Normal, Include, On, Off
Ch 6, Active, x24, Normal, Include, On, Off
Ch 7, Active, x24, Normal, Include, On, Off
Ch 8, Active, x24, Normal, Include, On, Off
Ch 9, Active, x24, Normal, Include, On, Off
Ch 10, Active, x24, Normal, Include, On, Off
Ch 11, Active, x24, Normal, Include, On, Off
Ch 12, Active, x24, Normal, Include, On, Off
Ch 13, Active, x24, Normal, Include, On, Off
Ch 14, Active, x24, Normal, Include, On, Off
Ch 15, Active, x24, Normal, Include, On, Off
Success writing channel 0
Return code: 200
Success writing channel 1
Return code: 200
Success writing channel 2
Return code: 200
Success writing channel 3
Return code: 200
Success writing channel 4
Return code: 200
Success writing channel 5
Return code: 200
Success writing channel 6
Return code: 200
Success writing channel 7
Return code: 200
Success writing channel 8
Return code: 200
Success writing channel 9
Return code: 200
Success writing channel 10
Return code: 200
Success writing channel 11
Return code: 200
Success writing channel 12
Return code: 200
Success writing channel 13
Return code: 200
Success writing channel 14
Return code: 200
Success writing channel 15
java.lang.RuntimeException: Waited 5000ms for: <10a68ca2, a9b662d>[count 2, qsz 0, owner <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1>] - <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1-FPSAWTAnimator#00-Timer2>
	at processing.opengl.PSurfaceJOGL$2.run(PSurfaceJOGL.java:410)
	at java.lang.Thread.run(Thread.java:748)
RuntimeException: Waited 5000ms for: <10a68ca2, a9b662d>[count 2, qsz 0, owner <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1>] - <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1-FPSAWTAnimator#00-Timer2>

```

Joe-Westra, 2019-11-07:
I found the function that is causing the initiation of the 5000ms timer by adding in  6 second delays throughout the suspect method: SoftwareSettings.load():

       
    ////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
    //                                                Load GUI Settings                                                       //
    ////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
    void load(String loadGUISettingsFileLocation) throws Exception {
       ...
       ...
```
} else {
            topNav.configSelector.configOptions.get(0).setString("Turn Expert Mode On");
            topNav.configSelector.configOptions.get(0).setColorNotPressed(topNav.configSelector.newGreen);
            println("LoadGUISettings: Expert Mode Off");
            expertModeToggle = false;
        }
println("*ADDING delay 2 to see if it breaks");
        delay(6000);
        println("*delay 2 complete");
```

        //Apply Framerate
        println("*LoadGUISettings: above framerate");
        
        frameRateCounter = loadFramerate;

        println("*ADDING delay 2.25 to see if it breaks");
        delay(6000);
        println("*delay 2.25 complete");

        println("*above loadApplyWidgetDropdownText");
        //Load and apply all of the settings that are in dropdown menus. It's a bit much, so it has it's own function below.
        
        
        println("*ADDING delay 2.5 to see if it breaks");
        delay(6000);
        println("*delay 2.5 complete");

        loadApplyWidgetDropdownText();
        
                
        println("*ADDING delay 3 to see if it breaks");
        delay(6000);
        println("*delay 3 complete");

                switch (frameRateCounter) {
            case 0:
                topNav.fpsButton.setString("24 fps");
                frameRate(24); //refresh rate ... this will slow automatically, if your processor can't handle the specified rate
                break;
            case 1:
                topNav.fpsButton.setString("30 fps");
                frameRate(30); //refresh rate ... this will slow automatically, if your processor can't handle the specified rate
                break;
            case 2:
                topNav.fpsButton.setString("45 fps");
                frameRate(45); //refresh rate ... this will slow automatically, if your processor can't handle the specified rate
                break;
            case 3:
                topNav.fpsButton.setString("60 fps");
                frameRate(60); //refresh rate ... this will slow automatically, if your processor can't handle the specified rate
                break;
        }

        println("*ADDING delay 3.5 to see if it breaks");
        delay(6000);
        println("*delay 3.5 complete");
          
        println("*above loadApplyTimeSeriesSettings");

        //Apply Time Series Settings Last!!!
        loadApplyTimeSeriesSettings();

        
Which provides the following output:

```
LoadGUISettings: Expert Mode Off
*ADDING delay 2 to see if it breaks
Hub: processCommand: success -- 24898
*delay 2 complete
*LoadGUISettings: above framerate
*ADDING delay 2.25 to see if it breaks
*delay 2.25 complete
*above loadApplyWidgetDropdownText
*ADDING delay 2.5 to see if it breaks
*delay 2.5 complete
Networking: Protocol mode set to UDP
Networking Data Types || show num dropdowns = 6
********** in screenresize within network widget
Apply UDP Networking Mode
*ADDING delay 3 to see if it breaks
*delay 3 complete
*ADDING delay 3.5 to see if it breaks
java.lang.RuntimeException: Waited 5000ms for: <516546dd, 32343ef4>[count 2, qsz 0, owner <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1>] - <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1-FPSAWTAnimator#00-Timer2>
	at processing.opengl.PSurfaceJOGL$2.run(PSurfaceJOGL.java:410)
	at java.lang.Thread.run(Thread.java:748)
RuntimeException: Waited 5000ms for: <516546dd, 32343ef4>[count 2, qsz 0, owner <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1>] - <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1-FPSAWTAnimator#00-Timer2>
```

Removing all but delay 3.5 crashes at the same spot.  The `loadApplyTimeSeriesSettings();` call is quite time consuming and causes the crash right near the completion of writing the channel updates (as seen in my previous post).

**Solution:** move the frameRate switch statement below the call to `loadApplyTimeSeriesSettings();` like so:

```          
        //Apply Time Series Settings Last!!!
        loadApplyTimeSeriesSettings();

        switch (frameRateCounter) {
            case 0:
                topNav.fpsButton.setString("24 fps");
                frameRate(24); //refresh rate ... this will slow automatically, if your processor can't handle the specified rate
                break;
            case 1:
                topNav.fpsButton.setString("30 fps");
                frameRate(30); //refresh rate ... this will slow automatically, if your processor can't handle the specified rate
                break;
            case 2:
                topNav.fpsButton.setString("45 fps");
                frameRate(45); //refresh rate ... this will slow automatically, if your processor can't handle the specified rate
                break;
            case 3:
                topNav.fpsButton.setString("60 fps");
                frameRate(60); //refresh rate ... this will slow automatically, if your processor can't handle the specified rate
                break;
        }
 ```
Hopefully this helps someone else!

retiutut, 2019-11-07:
@Joe-Westra Great detective work!!!

retiutut, 2019-11-13:
@BeeCoughs Can you try @Joe-Westra's Solution or do you need a standalone GUI to test?

BeeCoughs, 2019-11-13:
Unfortunately this doesn't help since my exception is thrown even before the "INITIALIZING SYSTEM" message. I tried moving the switch statement regardless, in case there was something I was overlooking, as well as adding a few of the print statements, but I am still not getting past "Setup is complete!" being printed to console.

retiutut, 2019-11-13:
@BeeCoughs 
- Please try 4.1.7-beta.0 from Processing/VisualStudioCode
- Go to Debugging.pde and change `boolean isVerbose = false;` to `true`
- Try to connect or use synthetic mode
- Share the full log file here

I want to see the full log trace for what's happening on your machine. 🔧 🛠

BeeCoughs, 2019-11-13:
I changed the isVerbose boolean to true, but it doesn't seem the console output is any longer. Is there supposed to be a log file generated somewhere that I'm not seeing?

I also wasn't able to try and connect my board since the GUI window is frozen completely blank.

> Console Log Started at Local Time: 2019-11-13_16-33-15
> Screen Resolution: 1920 X 1080
> Welcome to the Processing-based OpenBCI GUI!
> For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
> Launching application from local data dir
> OpenBCI_GUI: hubStart: OS Detected: Windows
> Local Version: 417, Latest Version: 416
> GUI is up to date!
> OpenBCI_GUI::Setup: Is RX mulitcast: false
> OpenBCI_GUI::Setup: Has RX joined multicast: false
> OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
> ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
> The file "C:\Users\Brian\Documents\OpenBCI_GUI\Settings\UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
> OpenBCI_GUI::Control Panel: Playback history file not found or other error.
> OpenBCI_GUI::Setup: Setup is complete!
> java.lang.RuntimeException: Waited 5000ms for: <48b1e45a, 4676133c>[count 3, qsz 1, owner <main-FPSAWTAnimator#00-Timer0>] - <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1>
> 	at processing.opengl.PSurfaceJOGL$2.run(PSurfaceJOGL.java:410)
> 	at java.lang.Thread.run(Thread.java:748)
> RuntimeException: Waited 5000ms for: <48b1e45a, 4676133c>[count 3, qsz 1, owner <main-FPSAWTAnimator#00-Timer0>] - <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1>
>

retiutut, 2019-11-13:
https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v4.1.7-beta.2
Is just now available,

If you want to try this also, you can fetch it from usptream/development. https://github.com/OpenBCI/OpenBCI_GUI/tree/development

retiutut, 2019-11-14:
```
OpenBCI_GUI::Setup: Setup is complete!
Instantiating hub object...
```
This is what is supposed to appear after "Setup is complete". Make sure you have the Hub in the data directory. Might be getting caught up because of of sudo/no sudo?

BeeCoughs, 2019-11-14:
Unfortunately it's still happening with the 4.1.7-beta.2 release.
The Hub is in the data directory and is version 2.1.0. It launches when I run the code and it shows up in the tray.
Not sure how to check if it's a sudo/no sudo issue.

retiutut, 2019-11-25:
Another thing to try would be using a different drawing library to render the GUI. We use P2D, or Processing 2D.

You can launch the GUI from Processing or VSCode and change the `size(win_x, win_y, P2D);` case below to JAVA2D or P3D.
```
void settings() {
    // If 1366x768, set GUI to 976x549 to fix #378 regarding some laptop resolutions
    // Later changed to 976x742 so users can access full control panel
    if (displayWidth == 1366 && displayHeight == 768) {
        size(976, 742, P2D);
    } else {
        //default 1024x768 resolution with 2D graphics
        size(win_x, win_y, P2D);
    }
}
```

I'm not saying the GUI will work entirely when you change this, but maybe it will shed light on the issue.

retiutut, 2020-01-21:
Just encountered the same error while testing 5.0.0alpha-2 development on my MacBook Air.

```I

BeeCoughs, 2020-01-31:
Hey, sorry that it's been a while since I had a chance to address this further.

Now running from VSCode with GUI version 4.1.7, Processing version 3.5.4, and changed the drawing library to P3D and it was the same story, however, when I switch to JAVA2D it didn't immediately crash after setup. However, as soon as I mouse over the GUI window (tested this a few times) I do get a new error:

```
Console Log Started at Local Time: 2020-01-31_13-53-58
Screen Resolution: 1920 X 1080
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Windows
TopNav: Internet Connection Not Available
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
The file "C:\Users\Brian\Documents\OpenBCI_GUI\Settings\UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
OpenBCI_GUI::Setup: Setup is complete!
Instantiating hub object...
Hub: CheckHubInit: The GUI is connected to the Hub!
ClassCastException: processing.awt.PSurfaceAWT$SmoothCanvas cannot be cast to com.jogamp.newt.opengl.GLWindow
```

I also tried with the new alpha release of ver 5.0.0 and got the same error. 

I did notice that the line from the console verifying the version/latest version is absent, but not sure if that means anything. Either way, hopefully that does shed some light on the issue! Thanks for keeping this open despite my slow correspondence here.

retiutut, 2020-01-31:
I'm reverting the version back to 4.2.0 and I might release using Processing 4: https://github.com/processing/processing4/releases/tag/processing-1270-4.0a1.

I've also changed the code so that a GUI version will always print. I'm aware that changing the drawing mode doesn't help. It shouldn't be hard to try Processing 4 from that release page.

retiutut, 2020-01-31:
@Joe-Westra When GUI v5 is released, it's possible we will not try to apply board settings **while** starting a session. This seems to slow down the Session Initialization anyways.

BeeCoughs, 2020-01-31:
Same problem with Processing v4.0a1 though the exception in the console output was a little more detailed:

```
Console Log Started at Local Time: 2020-01-31_15-35-44
Screen Resolution: 1920 X 1080
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Windows
TopNav: Internet Connection Not Available
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
The file "C:\Users\Brian\Documents\OpenBCI_GUI\Settings\UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
OpenBCI_GUI::Setup: Setup is complete!
Instantiating hub object...
Hub: CheckHubInit: The GUI is connected to the Hub!
ClassCastException: class processing.awt.PSurfaceAWT$SmoothCanvas cannot be cast to class com.jogamp.newt.opengl.GLWindow (processing.awt.PSurfaceAWT$SmoothCanvas and com.jogamp.newt.opengl.GLWindow are in unnamed module of loader 'app')
```

It also highlights line 1451 in the OpenBCI_GUI code, which I find strange, because the renderer is set as JAVA2D and that line is in an if statement conditional on the renderer being set as P2D or P3D (unless draw library != renderer in which case ignore this).

I have not tried on any other computers but should be able to either this weekend or early next week and get back to you.

BeeCoughs, 2020-02-03:
Just wanted to follow up: had a chance to try this on a Dell desktop today and it works fine. Wish I knew why my main machine has trouble, but at least I feel a little less crazy knowing that I was doing things correctly!

retiutut, 2020-02-03:
Just to recap:

>I'm running on an Intel i7-8700K @3.7GHz with 64GB of ram and a GeForce 2080Ti (with up-to-date drivers). Plenty of HDD space (more than a TB) It freezes on a blank (white) window.

> One bit in the readme that makes me wonder is the line that says "OpenGL acceleration is required". From what I can tell the NVidia driver I have supports OpenGL 4.6 but I suppose that doesn't [...]

To hopefully fix this, we have moved to Processing 4. I believe the devs are aware of JOGL issues and there is a doc stating it should be replaced.

retiutut, 2020-05-03:
It appears that some machine's graphics cards may not support OpenGL graphics in a functional way. Closing this issue for now, as it seems tied to certain PC hardware configurations.

closed 2020-05-03 by retiutut