OpenBCI/OpenBCI_GUI #590: v4.1.5 bug with WiFi shield 2.0.5 and Cyton 3.1.2: NullPointerException when drawing the ui

rivasd, 2019-09-09:
## Problem

Trying to stream data with the WiFi shield in direct mode on Windows 10 using the latest firmwares on the Cyton board and the shield fails. It looks like the connection went through (although there is a suspicious ``cyton firmware null`` in there at the end of the logs) but then the GUI fails when drawing the actual UI? Can confirm this does not happen using a bluetooth connection using the same board, bluetooth connection and streaming works as expected. Using Hub version 2.0.9

## Expected 

The UI would launch after connecting to the Wifi shield, instead it hangs because of the crash in the UI drawing thread

## Operating System and Version

Windows 10

## GUI Version

4.1.5

## Running standalone app

Both on standalone and Processing

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

Yes

## Console Log

```
Console Log Started at Local Time: 2019-09-09_13-32-15
Screen Resolution: 1920 X 1080
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Windows
java.net.UnknownHostException: github.com
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:184)
	at java.net.PlainSocketImpl.connect(PlainSocketImpl.java:172)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at sun.security.ssl.SSLSocketImpl.connect(SSLSocketImpl.java:666)
	at sun.security.ssl.BaseSSLSocketImpl.connect(BaseSSLSocketImpl.java:173)
	at sun.net.NetworkClient.doConnect(NetworkClient.java:180)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:463)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:558)
	at sun.net.www.protocol.https.HttpsClient.<init>(HttpsClient.java:264)
	at sun.net.www.protocol.https.HttpsClient.New(HttpsClient.java:367)
	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.getNewHttpClient(AbstractDelegateHttpsURLConnection.java:191)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1156)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:1050)
	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:177)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1564)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1492)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:347)
	at processing.core.PApplet.createInputRaw(PApplet.java:7145)
	at processing.core.PApplet.createInput(PApplet.java:7098)
	at processing.core.PApplet.loadStrings(PApplet.java:7561)
	at OpenBCI_GUI$TopNav.loadGUIVersionData(OpenBCI_GUI.java:15264)
	at OpenBCI_GUI$TopNav.<init>(OpenBCI_GUI.java:14897)
	at OpenBCI_GUI.delayedSetup(OpenBCI_GUI.java:465)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at processing.core.PApplet.method(PApplet.java:3860)
	at processing.core.PApplet$2.run(PApplet.java:3898)
The file "https://github.com/OpenBCI/OpenBCI_GUI/releases/latest" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
The file "C:\Users\rivas\Documents\OpenBCI_GUI\Settings\UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
OpenBCI_GUI::Setup: Setup is complete!
Instantiating hub object...
Hub: CheckHubInit: The GUI is connected to the Hub!
sourceList: click! 
setInterface: curInterface: -1
channel count set to 8
Setting Latency to 10000
Setting WiFi Internet Protocol to tcp
Setting WiFi style to dynamic
[DEFAULT]: Protocol Wifi Selected for Cyton
setInterface: curInterface: 2
Setting sample rate for Cyton to 1000Hz

sample rate set to: 1000
[DEFAULT]: Transfer Protocol set to wifi
refreshWifiList
wifiList: click! 
[DEFAULT]: Wifi Device Name = OpenBCI-D8A0
Settings: LogFileMaxDuration = 60 minutes
Calling initSystem()

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

OpenBCI_GUI: hub : Sent connect to Hub - Id: OpenBCI-D8A0 SampleRate: 1000Hz Latency: 10000ms
Index: 0, Size: 0
openBCI_GUI: haltSystem: Halting system for reconfiguration of settings...
Closing any open SD file. Writing 'j' to OpenBCI.
The file "C:\Users\rivas\Documents\OpenBCI_GUI\Settings\UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
Hub: processConnect: made it -- 64495 code: 200
Hub: setBoardType(): sending 'cyton -- 64495
Hub: processBoardType: success -- 64497
[SUCCESS]: The GUI is done initializing. Settings Loaded! Press "Start Data Stream" to start streaming! --  Cyton firmware null
java.lang.NullPointerException
	at OpenBCI_GUI.systemUpdate(OpenBCI_GUI.java:1281)
	at OpenBCI_GUI.draw(OpenBCI_GUI.java:577)
	at processing.core.PApplet.handleDraw(PApplet.java:2475)
	at processing.opengl.PSurfaceJOGL$DrawListener.display(PSurfaceJOGL.java:866)
	at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
	at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
	at jogamp.opengl.GLAutoDrawableBase$2.run(GLAutoDrawableBase.java:443)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
	at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
	at com.jogamp.newt.opengl.GLWindow.display(GLWindow.java:759)
	at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:81)
	at com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:452)
	at com.jogamp.opengl.util.FPSAnimator$MainTask.run(FPSAnimator.java:178)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
```

wjcroft, 2019-09-10:
Daniel, hi.

The current reality at OpenBCI, seems to be that Wifi Shield issues are being assigned "low priority". This is for a couple reasons: (1) the author of the wifi code, AJ Keller, has founded his own company (Neurocity), and no longer is working for or even consulting for OpenBCI. He has stated to us that he has no free time to look into such issues. (2) Given this situation and the complexity of the product, the Wifi Shield on the Shop page has been labeled with a disclaimer, 

> IMPORTANT: the WiFi shield is known to have reliability issues across different computer configurations. Using it effectively requires advanced technical skills and programming knowledge. Please proceed at your own risk. OpenBCI cannot guarantee that the device will work with your configuration. 

Have you tried various other configurations in an attempt to see if a workaround might exist? I do see that you are using TCP mode, 1000Hz sample rate, and Wifi Direct. Have you tried UDPx3, 500Hz, Wifi through the router, etc.? It does appear that the null pointer exception on this issue is a bug in the GUI, but other settings may influence this.

UDPx3 is known to be more reliable than TCP. Is your application an EEG experiment? 500 Hz is generally fine for EEG. Although Wifi through a router is slower than Direct, it is also possible to configure small inexpensive hotspots to act as routers, etc.

Let us know if you find a workaround. Mentioning @retiutut @conorrussomanno @produceconsumerobot 

Regards, William

wjcroft, 2019-09-10:
Related forum post is here, https://openbci.com/forum/index.php?p=/discussion/2293/impedance-commands-not-working-through-wifi-shield-api-cyton-daisy

retiutut, 2019-09-10:
While this is low priority, it is a crash. I can try to look into this for the next GUI release.

retiutut, 2019-09-12:
Unable to reproduce this issue of the NullPointerException, but you should be able to make it into the session now.

Please do `git fetch --all` and `git merge upstream/development`.

retiutut, 2019-10-01:
I think I saw this while running 4.1.6 on Windows. I reopened the GUI and I was able to connect without a problem

retiutut, 2019-10-13:
@rivasd What's the status on this issue with GUI 4.1.6?

closed 2019-10-18 by retiutut