OpenBCI/OpenBCI_GUI #591: GUI freezes on MacOS when clicking 'SELECT PLAYBACK FILE'

bacpac, 2019-09-10:
## Problem

When trying to use 'PLAYBACK (from file)', when actually clicking on 'SELECT PLAYBACK FILE', the GUI crashes before showing the file selector window. Recent files are not shown in the 'PLAYBACK HISTORY' either.

I've updated from 4.1.2 to 4.1.5. When downgrading the file selector works again (and recent files are shown as well).

## Expected 

I'd expect the file selector window to pop-up.

## Operating System and Version

macOS 10.14.6

## GUI Version

4.1.5 (it still works on 4.1.2)

## Running standalone app

Stand alone app

## Type of OpenBCI Board

None, just data from a text file

## Are you using a WiFi Shield?

No

## Console Log

-

retiutut, 2019-09-10:
Please post the Console Log file and the Recording file.

retiutut, 2019-09-10:
https://github.com/OpenBCI/OpenBCI_GUI/issues/487

retiutut, 2019-09-10:
Please try updating your java runtime environment. https://java.com/en/download/mac_download.jsp

closed 2019-09-10 by retiutut

retiutut, 2019-09-10:
Just added this info to the Readme for this Repo.

https://github.com/OpenBCI/OpenBCI_GUI/blob/master/README.md#troubleshooting

retiutut, 2019-09-10:
@bacpac Or.... you can try the updated package I am about to upload for 4.1.5 that explicitly includes a Java runtime in the the .app

retiutut, 2019-09-10:
@bacpac Please try the updated GUI download on the releases page before installing Java. It should work. If you can, let me know this works for you.

bacpac, 2019-09-17:
@retiutut 
Apologies for my late response; but it works again!

As you've requested, I've refrained from manually updating Java and downloaded the latest release (4.1.5) again. I can confirm that I'm able to select a file and play it. Thank you for the quick support :)