OpenBCI/OpenBCI_GUI #592: Networking Widget LSL streams not working for other data types than EEG

lopezvaa, 2019-09-11:
## Problem

The LSL streaming of Accelerometer of Digital Channels does not work. The streams come out empty after being recorded with Lab Recorder for LSL streams.

The stream channels are visible and the info for each outlet is recorded correctly. But there is no data available in the stream

## Expected 

According to the Network Widget Guide, accelerometer or analogue or  digital inputs should be possible to stream with the LSL protocol in the latest release 4.1.5 beta3. 

The EEG channels are recorded as expected and can be seen in the figure below.

[image]

When recorded with and LSL stream recorder like Lab Recorder, I expect to see an LSL stream with accelerometer or auxiliary input just as the EEG timeseries above. Instead those streams are empty, see below figure.

[image]
 

## Operating System and Version

Windows 10, 64-bit

## GUI Version

4.1.5 beta3 release

## Running standalone app

Problem was found in both stand alone and processing versions

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

retiutut, 2019-09-11:
@lopezvaa Please tell me exactly what I need to receive LSL just like you are here. Step by step.

I have already reproduced the other issue you have just opened, and I need help replicating this issue also. Thanks!

lopezvaa, 2019-09-11:
Sure @Richard Waltman<mailto:notifications@github.com> ,

I’m away from my workstation for the evening but I can give you more details tomorrow.  In short:
1) set up OpenBCI GUI on a new session via dongle, select the widgets timeseries, networking and digital read on

2) set the networking widget to stream via LSL protocol 2 active streams (obci_eeg1, timeseries, type EEG, 8 chn; obci_eeg2, accel/aux, type DIGITAL, 5 chn)

3) start the data stream on the session and start the LSL stream

4)in a different software (I use matlab and MatLSL-lib) create a simple LSL consumer script with 2 inlets, one of obci_eeg1 and one for obci_eeg2. Pull one or more chunks from each LSL inlet and display the data received.

5) you will see that the obci_eeg1 data chunk has 8 chns and however many samples, representing the EEG data. But the second stream obci_eeg2 is empty even when it was detected as an active stream.

6) repeat steps 1 to 5 but now invert the order of the streams in step 2 (obci_eeg1 to be the aux data and obci_eeg2 the EEG data)

7) you will now see that your data pull receives the 5 chns of DIGITAL inputs but the EEG stream is empty.

That will replicate the error. I can elaborate with screen dumps tomorrow for more clarity if needed.

Thanks for looking into the issues
Kind regards
Alejandro

Alejandro Lopez
M.Sc, Ph.D
Research Engineer, Cognitive Hearing Science

Eriksholm Research Centre
Rørtangvej 20
3070 Snekkersten
Denmark
Direct: +45 4829 8913
IPT: *454913
Mobile: +45 6022 9862
Email: aova@eriksholm.com
Website: www.eriksholm.com

Eriksholm pursues audiological discoveries with the potential to significantly enhance end-user benefits in future hearing care.
We work in close collaboration with academic research institutions, clinicians and end-users, and being part of Oticon ensures that our scientific insights are applied in solutions that empower people with hearing loss.

lopezvaa, 2019-09-18:
I think the problem may be in the checkForData function of the Networking Widget

    Boolean checkForData() {
        if (this.dataType.equals("TimeSeries")) {
            return dataProcessing.newDataToSend;
        } else if (this.dataType.equals("FFT")) {
            return dataProcessing.newDataToSend;
        } else if (this.dataType.equals("EMG")) {
            return dataProcessing.newDataToSend;
        } else if (this.dataType.equals("BandPower")) {
            return dataProcessing.newDataToSend;
        } else if (this.dataType.equals("Accel/Aux")) {
            return dataProcessing.newDataToSend;
        } else if (this.dataType.equals("Focus")) {
            return dataProcessing.newDataToSend;
        } else if (this.dataType.equals("Pulse")) {
            return dataProcessing.newDataToSend;
        } else if (this.dataType.equals("SSVEP")) {
            return dataProcessing.newDataToSend;
        }
        return false;
    }

When you have simultaneous streams running dataProcessing.newDataToSend seems to be dominated by the first stream data type. 

But I have not been able to find a solution to it. Yet, in just getting the hang of processing.

lopezvaa, 2019-09-18:
Bingo! that seems to be it. but I still haven't found a solution, I just hardcoded a fix for the moment. 
here a snapshot of the result.

[image]

retiutut, 2019-09-21:
@lopezvaa Can you make a Pull Request so we can start coding/merging a solution to this issue?

retiutut, 2019-10-01:
I have just replicated this issue while trying to receive data using a pylsl:

```
"""Example program to demonstrate how to read a multi-channel time-series
from LSL in a chunk-by-chunk manner (which is more efficient)."""

from pylsl import StreamInlet, resolve_stream

numStreams = 2
# first resolve an EEG stream on the lab network
print("looking for an EEG stream...")
stream1 = resolve_stream('name', 'obci_eeg1')
stream2 = resolve_stream('name', 'obci_eeg2')

# create a new inlet to read from the stream
inlet = StreamInlet(stream1[0])
inlet2 = StreamInlet(stream2[0])

while True:
    for i in range(numStreams):
        # get a new sample (you can also omit the timestamp part if you're not
        # interested in it)
        if i == 0:
            chunk, timestamps = inlet.pull_chunk()
            #if timestamps:
                #print("Stream", i + 1, " == ", chunk)
        elif i == 1:
            chunk, timestamps = inlet2.pull_chunk()
            if timestamps:
                    print("Stream", i + 1, " == ", chunk)
```

retiutut, 2019-10-01:
[image]

closed 2019-10-01 by retiutut

retiutut, 2019-10-01:
@lopezvaa Try https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v4.1.6 !!! 🔥

lopezvaa, 2019-10-08:
> @lopezvaa Try https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v4.1.6 !!! 🔥

Thanks @retiutut I'm jsut back from a conference, so I'll get back to this now and try the suggested version! Awesome!