OpenBCI/OpenBCI_GUI #593: Networking Widget stop button not working and causes crashing of the GUI

lopezvaa, 2019-09-11:
## Problem

When starting data streams through the Networking Widget after pressing the start button on the streamer, the data stream starts as expected, the button then turns into a red 'STOP' button but the stream does not stop. 

This causes the widget to get stuck on its current configuration and in order to change the streams, the session has to be stopped. However this causes the GUI to crash and fails to start a new session.

## Expected 

I would expect the streams to stop when the 'STOP' button is pressed and the widget can be modified and started again. If there is only one networking chance per session then the widget should not crash the GUI after the session has been stopped

[image]

## Operating System and Version

Windows 10, 64-bit

## GUI Version

4.1.5 beta 3

## Running standalone app

Both stand alone and processing are faulty

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

## Console Log

Console Log Started at Local Time: 2019-09-11_12-52-47
Screen Resolution: 1920 X 1080
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Windows
Local Version: 415, Latest Version: 415
GUI is up to date!
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
The file "C:\Users\aova\Documents\OpenBCI_GUI\Settings\UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
OpenBCI_GUI::Setup: Setup is complete!
Instantiating hub object...
Hub: CheckHubInit: The GUI is connected to the Hub!
sourceList: click! 
setInterface: curInterface: -1
channel count set to 8
Setting Latency to 10000
Setting WiFi Internet Protocol to tcp
Setting WiFi style to dynamic
[DEFAULT]: Protocol Serial Selected for Cyton
setInterface: curInterface: 0
Setting sample rate for Cyton to 250Hz

sample rate set to: 250
[DEFAULT]: Transfer Protocol set to serial
serialList: click! 
[DEFAULT]: OpenBCI Port Name = COM10
Settings: LogFileMaxDuration = 60 minutes
Calling initSystem()

=================================================
||             INITIALIZING SYSTEM             ||
[... 138 lines cut ...]
Ch 3, Active, x24, Normal, Include, On, Off
Ch 4, Active, x24, Normal, Include, On, Off
Ch 5, Active, x24, Normal, Include, On, Off
Ch 6, Active, x24, Normal, Include, On, Off
Ch 7, Active, x24, Normal, Include, On, Off
Return code: 200
Success writing channel 0
Return code: 200
Success writing channel 1
Return code: 200
Success writing channel 2
Return code: 200
Success writing channel 3
Return code: 200
Success writing channel 4
Return code: 200
Success writing channel 5
Return code: 200
Success writing channel 6
Return code: 200
Success writing channel 7
java.lang.RuntimeException: Waited 5000ms for: <25f020db, 2b37da15>[count 2, qsz 0, owner <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1-FPSAWTAnimator#00-Timer2>] - <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1-FPSAWTAnimator#00-Timer2-FPSAWTAnimator#00-Timer3>
	at processing.opengl.PSurfaceJOGL$2.run(PSurfaceJOGL.java:410)
	at java.lang.Thread.run(Thread.java:748)
RuntimeException: Waited 5000ms for: <25f020db, 2b37da15>[count 2, qsz 0, owner <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1-FPSAWTAnimator#00-Timer2>] - <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1-FPSAWTAnimator#00-Timer2-FPSAWTAnimator#00-Timer3>

retiutut, 2019-09-11:
Really had to rename this because the stop/start button is in fact working. Not surprised if LSL has bugs because it was completely non-functional for a very long time.

We are only finding these bugs out now because LSL is working.

retiutut, 2019-09-11:
I can see now that the Start/Stop button is broken in the networking widget.

retiutut, 2019-09-16:
The code for stopping the networking stream got removed accidentally when adding another button. Easy fix. 
https://github.com/OpenBCI/OpenBCI_GUI/commit/bc0e085b28fe1bbe433de76fa26fd7878f96c365#diff-32370e0f7a6c1e0b4ea80be34d4d7473

closed 2019-09-16 by retiutut