OpenBCI/OpenBCI_GUI #597: serial dongle->cyton communication lost from OpenBCI_GUI_4.1.2 to newer versions

djan4848, 2019-09-14:
## Problem

In Linux Ubuntu 18.04: , OpenBCI_GUI works fine using Cyton (V3.32) + Daisy (V3) with serial Dongle on OpenBCI_GUI 4.1.2 but doesn't work with later versions (verified with versions 4.1.3 to 4.1.5) 
## Expected 
a working communication dongle->cyton with possibility to stream data as in OpenBCI_GUI version 4.1.2

## Operating System and Version
Linux ubuntu 18.04
(Linux djan-Latitude-E5520 4.4.0-139-generic #165-Ubuntu SMP Wed Oct 24 10:58:50 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux)

## GUI Version

Works fine 4.1.2   doesn't work 4.1.3 to 4.1.5

## Running standalone app

standalone linux

## Type of OpenBCI Board

Cyton+Daisy

## Are you using a WiFi Shield?

No

## Console Log

WHEN IT DOESN'T WORK (version 4.1.3):
 getting following message: "
_Disconnecting serial...
[ERROR]: Failed to connect to data source. Check that the device is powered on and in range. Also, try pressing AUTOSCAN."_
This are the traces when it doesn't work:
jan@djan-Latitude-E5520:~/Investigaciones/openBCI/OpenBCI_GUI_4.1.3$ ./OpenBCI_GUI
Console Log Started at Local Time: 2019-09-14_13-57-19
Screen Resolution: 1920 X 1080
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://docs.openbci.com/OpenBCI%20Software/
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Linux
Local Version: 413, Latest Version: 415
GUI needs to be updated. Download at https://github.com/OpenBCI/OpenBCI_GUI/releases/latest
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
The file "/home/djan/Documents/OpenBCI_GUI/Settings/UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
OpenBCI_GUI::Setup: Setup is complete!
Instantiating hub object...
Hub: clientEvent: handshake complete
sourceList: click! 
setInterface: curInterface: -1
channel count set to 8
Setting Latency to 10000
Setting WiFi Internet Protocol to tcp
Setting WiFi style to dynamic
[DEFAULT]: Protocol Serial Selected for Cyton
setInterface: curInterface: 0
Setting sample rate for Cyton to 250Hz

sample rate set to: 250
[DEFAULT]: Transfer Protocol set to serial
serialList: click! 
[DEFAULT]: OpenBCI Port Name = /dev/ttyUSB0
channel count set to 16
Settings: LogFileMaxDuration = 60 minutes
Calling initSystem()

=================================================
[... 116 lines cut ...]
DEBUG: start data transfer
Cyton: startDataTransfer(): writing 'b' to the serial port...
Hub: sendCommand(char): sending 'b'
Hub: processCommand: success -- 36657
openBCI_GUI: stopButton was pressed...stopping data transfer...
OpenBCI_GUI: stopRunning: stop running...
[DEFAULT]: Data stream stopped.
Cyton: startDataTransfer(): writing 's' to the serial port...
Hub: sendCommand(char): sending 's'
Hub: processCommand: success -- 39104
-- UDP session started at Sat Sep 14 13:59:55 CEST 2019 --
-- bound socket to host:127.0.0.1, port: 51000 --
[19-09-14 14:00:45.882 +0200] close socket < port:51000, address:127.0.0.1 >

Client SocketException: Socket closed
SHUTDOWN HOOK
X11Util.Display: Shutdown (JVM shutdown: true, open (no close attempt): 2/2, reusable (open, marked uncloseable): 0, pending (open in creation order): 2)
X11Util: Open X11 Display Connections: 2
X11Util: Open[0]: NamedX11Display[:0, 0x7fedcc003340, refCount 1, unCloseable false]
X11Util: Open[1]: NamedX11Display[:0, 0x7fedcc011310, refCount 1, unCloseable false]
Killed: 20237 pts/0    00:00:01 OpenBCIHub
Killed: 20239 pts/0    00:00:00 OpenBCIHub
Killed: 20271 pts/0    00:00:00 OpenBCIHub
SHUTDOWN HUB
3 active widgets saved!

retiutut, 2019-09-16:
In both cases it fails to get your Cyton firmware, that's why it abandons the System initialization in 4.1.2+.

retiutut, 2019-09-16:
@djan4848 I can either allow you to start the GUI without having the firmware read. Can you confirm the Cyton firmware is 3.1.1 or 3.1.2?

djan4848, 2019-09-17:
Hi Richard,

thank you for your answer.

this is the output get through the serial port when I talk to cyton 
without the GUI:

?
Board ADS Registers
ADS_ID, 0x00, 0x3E, 0, 0, 1, 1, 1, 1, 1, 0
CONFIG1, 0x01, 0xB6, 1, 0, 1, 1, 0, 1, 1, 0
CONFIG2, 0x02, 0xC0, 1, 1, 0, 0, 0, 0, 0, 0
CONFIG3, 0x03, 0xEC, 1, 1, 1, 0, 1, 1, 0, 0
LOFF, 0x04, 0x02, 0, 0, 0, 0, 0, 0, 1, 0
CH1SET, 0x05, 0x68, 0, 1, 1, 0, 1, 0, 0, 0
CH2SET, 0x06, 0x68, 0, 1, 1, 0, 1, 0, 0, 0
CH3SET, 0x07, 0x68, 0, 1, 1, 0, 1, 0, 0, 0
CH4SET, 0x08, 0x68, 0, 1, 1, 0, 1, 0, 0, 0
CH5SET, 0x09, 0x68, 0, 1, 1, 0, 1, 0, 0, 0
CH6SET, 0x0A, 0x68, 0, 1, 1, 0, 1, 0, 0, 0
CH7SET, 0x0B, 0x68, 0, 1, 1, 0, 1, 0, 0, 0
CH8SET, 0x0C, 0x68, 0, 1, 1, 0, 1, 0, 0, 0
BIAS_SENSP, 0x0D, 0xFF, 1, 1, 1, 1, 1, 1, 1, 1
BIAS_SENSN, 0x0E, 0xFF, 1, 1, 1, 1, 1, 1, 1, 1
LOFF_SENSP, 0x0F, 0x00, 0, 0, 0, 0, 0, 0, 0, 0
LOFF_SENSN, 0x10, 0x00, 0, 0, 0, 0, 0, 0, 0, 0
LOFF_FLIP, 0x11, 0x00, 0, 0, 0, 0, 0, 0, 0, 0
LOFF_STATP, 0x12, 0x00, 0, 0, 0, 0, 0, 0, 0, 0
LOFF_STATN, 0x13, 0x00, 0, 0, 0, 0, 0, 0, 0, 0
GPIO, 0x14, 0x0F, 0, 0, 0, 0, 1, 1, 1, 1
MISC1, 0x15, 0x00, 0, 0, 0, 0, 0, 0, 0, 0
MISC2, 0x16, 0x00, 0, 0, 0, 0, 0, 0, 0, 0
CONFIG4, 0x17, 0x00, 0, 0, 0, 0, 0, 0, 0, 0

Daisy ADS Registers
ADS_ID, 0x00, 0x3E, 0, 0, 1, 1, 1, 1, 1, 0
CONFIG1, 0x01, 0x96, 1, 0, 0, 1, 0, 1, 1, 0
CONFIG2, 0x02, 0xC0, 1, 1, 0, 0, 0, 0, 0, 0
CONFIG3, 0x03, 0xEC, 1, 1, 1, 0, 1, 1, 0, 0
LOFF, 0x04, 0x02, 0, 0, 0, 0, 0, 0, 1, 0
CH1SET, 0x05, 0x68, 0, 1, 1, 0, 1, 0, 0, 0
CH2SET, 0x06, 0x68, 0, 1, 1, 0, 1, 0, 0, 0
CH3SET, 0x07, 0x68, 0, 1, 1, 0, 1, 0, 0, 0
CH4SET, 0x08, 0x68, 0, 1, 1, 0, 1, 0, 0, 0
CH5SET, 0x09, 0x68, 0, 1, 1, 0, 1, 0, 0, 0
CH6SET, 0x0A, 0x68, 0, 1, 1, 0, 1, 0, 0, 0
CH7SET, 0x0B, 0x68, 0, 1, 1, 0, 1, 0, 0, 0
CH8SET, 0x0C, 0x68, 0, 1, 1, 0, 1, 0, 0, 0
BIAS_SENSP, 0x0D, 0xFF, 1, 1, 1, 1, 1, 1, 1, 1
BIAS_SENSN, 0x0E, 0xFF, 1, 1, 1, 1, 1, 1, 1, 1
LOFF_SENSP, 0x0F, 0x00, 0, 0, 0, 0, 0, 0, 0, 0
LOFF_SENSN, 0x10, 0x00, 0, 0, 0, 0, 0, 0, 0, 0
LOFF_FLIP, 0x11, 0x00, 0, 0, 0, 0, 0, 0, 0, 0
LOFF_STATP, 0x12, 0x00, 0, 0, 0, 0, 0, 0, 0, 0
LOFF_STATN, 0x13, 0x00, 0, 0, 0, 0, 0, 0, 0, 0
GPIO, 0x14, 0x0F, 0, 0, 0, 0, 1, 1, 1, 1
MISC1, 0x15, 0x00, 0, 0, 0, 0, 0, 0, 0, 0
MISC2, 0x16, 0x00, 0, 0, 0, 0, 0, 0, 0, 0
CONFIG4, 0x17, 0x00, 0, 0, 0, 0, 0, 0, 0, 0

LIS3DH Registers
0x07    0
0x08    0
0x09    0
0x0A    0
0x0B    0
0x0C    0
0x0D    0
0x0E    0
0x0F    33

0x1F    0
0x20    8
0x21    0
0x22    0
0x23    18
0x24    0
0x25    0
0x26    0
0x27    0
0x28    0
0x29    0
0x2A    0
0x2B    0
0x2C    0
0x2D    0
0x2E    0
0x2F    20
0x30    0
0x31    0
0x32    0
0x33    0

0x38    0
0x39    0
0x3A    0
0x3B    0
0x3C    0
0x3D    0
$$$
OpenBCI V3 16 channel
On Board ADS1299 Device ID: 0x3E
On Daisy ADS1299 Device ID: 0x3E
LIS3DH Device ID: 0x33
$$$

Hope this help to clarify the problem.

Regards

Damian

retiutut, 2019-09-17:
@djan4848 When and where did you buy the Cyton+Daisy from?

djan4848, 2019-09-17:
I buyed  a R&D 16 channels kit on openBCI Shop (order #1825) jan 04, 
2016, then I buyed a new cyton (order#3109) mar 1, 2017

I hope this help.

Regards

Damian Jan

retiutut, 2019-09-18:
if cyton is running older than 2.x.x. firmware you will need to update, by following this tutorial: https://docs.openbci.com/docs/02Cyton/CytonProgram#firmware-version-1xx-2014-fall-2016

retiutut, 2019-09-19:
https://docs.openbci.com/docs/02Cyton/CytonSDK#cyton-command-protocol-overview

> OpenBCI V3 16 channel
> On Board ADS1299 Device ID: 0x3E
> On Daisy ADS1299 Device ID: 0x3E
> LIS3DH Device ID: 0x33
> $$$

This means you are actually running 1.x.x firmware. There is no problem with the GUI, but it is time to update the Cyton firmware using this tutorial: https://docs.openbci.com/docs/02Cyton/CytonProgram#firmware-version-1xx-2014-fall-2016

I think this will resolve the issue. 😄

bmartin427, 2019-09-19:
Would it be fair to re-classify this bug as "GUI error message when using old Cyton firmware could be clearer"?

retiutut, 2019-09-19:
Init messages are complicated. But I can try to let the users know why the init process quits in this case.

retiutut, 2019-09-19:
This whole bit here in InterfaceHub.pde doesn't seem right now either, because I think initAndShowGUI() happens for both Ganglion and Cyton. Let me see...

```
        //This success message appears in Cyton mode
        if (settings.loadErrorCytonEvent == true) {
            outputError("Connection Error: Failed to apply channel settings to Cyton.");
        } else {
            outputSuccess("The GUI is done initializing. " + settingsString + "Press \"Start Data Stream\" to start streaming! -- " + firmwareString);
        }
```

retiutut, 2019-09-20:
```
outputError("Check that the device is powered on and in range. Also, try AUTOSCAN. Otherwise, Cyton firmware is out of date.");
```

bmartin427, 2019-09-20:
Sure, that's probably a significant improvement.  Better still would be to print a more specific error message in only that case; but, the current control flow in `initSystem()` makes that tricky.

Maybe what you've got in the generic `abandonInit` case, plus a `println()` after the `getFirmwareVersion()` comparison fails with a more specific clue?

retiutut, 2019-09-20:
This is about as specific as we can be for now. One sure way to get this error scenario is to have the dongle plugged in and the Cyton is powered off. We may be able to improve the connection process at a later date using Java libraries that are a work in progress.

Back to the original issue, all @djan4848 needs to do is update Cyton firmware and it should work. Waiting to hear back so I can close this issue.

djan4848, 2019-09-22:
Hi Richard,

I have already updated the firmware, now the GUI pass the 
initialization, but when I start streaming the GUI doesn't show 
anything. On the log I get this message:

/Hub: processCommand: success -- 221975//
//Hub: apparent sampleIndex jump from Serial data: 0 to 104.  Keeping 
packet. (1)//
//numPacketsDropped = 103/

The output file only contents the header:

/%OpenBCI Raw EEG Data//
//%Number of channels = 16//
//%Sample Rate = 125.0 Hz//
//%First Column = SampleIndex//
//%Last Column = Timestamp//
//%Other Columns = EEG data in microvolts followed by Accel Data (in G) 
interleaved with Aux Data/

The Dongle red light is fixed in red, no blinking.

The behavior is the same with version GUI version 410, even if before 
the update it used to work fine.

This is the log of version 4.1.5:

/djan@Base:~/DATA/Investigaciones/openBCI/OpenBCI_GUI$ ./OpenBCI_GUI//
//Console Log Started at Local Time: 2019-09-22_20-09-58//
//Screen Resolution: 1920 X 1080//
//Welcome to the Processing-based OpenBCI GUI!//
//For more information, please visit: 
https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs//
//Launching application from local data dir//
//OpenBCI_GUI: hubStart: OS Detected: Linux//
//Local Version: 415, Latest Version: 415//
//GUI is up to date!//
//OpenBCI_GUI::Setup: Is RX mulitcast: false//
//OpenBCI_GUI::Setup: Has RX joined multicast: false//
//OpenBCI_GUI::Setup: Sample Data exists in Documents folder.//
//ControlP5 2.2.6 infos, comments, questions at 
http://www.sojamo.de/libraries/controlP5//
//The file 
"/home/djan/Documents/OpenBCI_GUI/Settings/UserPlaybackHistory.json" is 
missing or inaccessible, make sure the URL is valid or that the file has 
been added to your sketch and is readable.//
//OpenBCI_GUI::Control Panel: Playback history file not found or other 
error.//
//OpenBCI_GUI::Setup: Setup is complete!//
//Instantiating hub object...//
//Hub: CheckHubInit: The GUI is connected to the Hub!//
//sourceList: click! //
//setInterface: curInterface: -1//
//channel count set to 8//
//Setting Latency to 10000//
//Setting WiFi Internet Protocol to tcp//
//Setting WiFi style to dynamic//
//[DEFAULT]: Protocol Serial Selected for Cyton//
//setInterface: curInterface: 0//
//Setting sample rate for Cyton to 250Hz//
//
//
//sample rate set to: 250//
//[DEFAULT]: Transfer Protocol set to serial//
//serialList: click! //
//[DEFAULT]: OpenBCI Port Name = /dev/ttyUSB0//
//channel count set to 16//
//Settings: LogFileMaxDuration = 60 minutes//
//Calling initSystem()//
//
//
//=================================================//
//||             INITIALIZING SYSTEM             ||//
//=================================================//
//
//"Arial" is not available, so another font will be used. Use 
PFont.list() to show available fonts.//
//Hub: processConnect: made it -- 205265 code: 200//
//Hub: setBoardType(): sending 'daisy -- 205266//
//Hub: processBoardType: success -- 205267//
//[SUCCESS]: The GUI is done initializing. Settings Loaded! Press "Start 
Data Stream" to start streaming! --  Cyton firmware v3.1.2//
//InitSettings: Saving Default Settings to file!//
//SoftwareSettings: 3 active widgets saved!//
//LoadGUISettings: Applied Widget 0 to Container 4//
//LoadGUISettings: Applied Widget 1 to Container 9//
//LoadGUISettings: Applied Widget 5 to Container 3//
//Hub: sendCommand(String): sending '/0'//
//Cyton: setBoardMode to :DEFAULT//
//LoadGUISettings: Expert Mode Off//
//Hub: processCommand: success -- 205825//
//Networking: Protocol mode set to LSL//
//Networking Data Types || show num dropdowns = 13//
//Apply LSL Networking Mode//
//Ch 0, Active, x24, Normal, Include, On, Off//
//Ch 1, Active, x24, Normal, Include, On, Off//
//Ch 2, Active, x24, Normal, Include, On, Off//
//Ch 3, Active, x24, Normal, Include, On, Off//
//Ch 4, Active, x24, Normal, Include, On, Off//
//Ch 5, Active, x24, Normal, Include, On, Off//
//Ch 6, Active, x24, Normal, Include, On, Off//
//Ch 7, Active, x24, Normal, Include, On, Off//
//Ch 8, Active, x24, Normal, Include, On, Off//
//Ch 9, Active, x24, Normal, Include, On, Off//
//Ch 10, Active, x24, Normal, Include, On, Off//
//Ch 11, Active, x24, Normal, Include, On, Off//
//Ch 12, Active, x24, Normal, Include, On, Off//
//Ch 13, Active, x24, Normal, Include, On, Off//
//Ch 14, Active, x24, Normal, Include, On, Off//
//Ch 15, Active, x24, Normal, Include, On, Off//
//Success writing channel 0//
//Return code: 200//
//Success writing channel 1//
//Return code: 200//
//Success writing channel 2//
//Return code: 200//
//Success writing channel 3//
//Return code: 200//
//Success writing channel 4//
//Return code: 200//
//Success writing channel 5//
//Return code: 200//
//Success writing channel 6//
//Return code: 200//
//Success writing channel 7//
//Return code: 200//
//Success writing channel 8//
//Return code: 200//
//Success writing channel 9//
//Return code: 200//
//Success writing channel 10//
//Return code: 200//
//Success writing channel 11//
//Return code: 200//
//Success writing channel 12//
//Return code: 200//
//Success writing channel 13//
//Return code: 200//
//Success writing channel 14//
//Return code: 200//
//Success writing channel 15//
//Headplot is active: Redrawing//
//[DEFAULT]: Data stream started.//
//Cyton: startDataTransfer(): writing 'b' to the serial port...//
//Hub: sendCommand(char): sending 'b'//
//OpenBCI_GUI: openNewLogFile: opened ODF output file: 
/home/djan/Documents/OpenBCI_GUI/Recordings/OpenBCISession_2019-09-22_20-10-02/OpenBCI-RAW-2019-09-22_20-13-37.txt//
//Hub: processCommand: success -- 221975//
//Hub: apparent sampleIndex jump from Serial data: 0 to 104.  Keeping 
packet. (1)//
//numPacketsDropped = 103/

retiutut, 2019-09-22:
@wjcroft Does @djan4848 need to update the Cyton Dongle also?

The GUI does read the firmware as 3.1.2, but apparently drops 100+ packets when start button is pressed.

retiutut, 2019-09-22:
@djan4848 Please try a few times. Let me know if it does this exact thing EVERY time.

djan4848, 2019-09-22:
Not allways, but it doesn't show anything on the GUI, and the log files 
only have one line of data when it arrives to keep one packet.

/Hub: processCommand: success -- 32904//
//Hub: apparent sampleIndex jump from Serial data: 0 to  64. Keeping 
packet. (1)//
//numPacketsDropped = 63//
/

/Headplot is active: Redrawing//
//[DEFAULT]: Data stream started.//
//Cyton: startDataTransfer(): writing 'b' to the serial port...//
//Hub: sendCommand(char): sending 'b'//
//OpenBCI_GUI: openNewLogFile: opened ODF output file: 
/home/djan/Documents/OpenBCI_GUI/Recordings/OpenBCISession_2019-09-22_23-28-38/OpenBCI-RAW-2019-09-22_23-29-21.txt//
//Hub: processCommand: success -- 48991//
//[DEFAULT]: Data stream stopped.//
//Cyton: startDataTransfer(): writing 's' to the serial port...//
//Hub: sendCommand(char): sending 's'//
//Hub: processCommand: success -- 88165//
//[DEFAULT]: Data stream started.//
//Cyton: startDataTransfer(): writing 'b' to the serial port...//
//Hub: sendCommand(char): sending 'b'//
//OpenBCI_GUI: openNewLogFile: opened ODF output file: 
/home/djan/Documents/OpenBCI_GUI/Recordings/OpenBCISession_2019-09-22_23-28-38/OpenBCI-RAW-2019-09-22_23-30-03.txt//
//Hub: processCommand: success -- 90930//
//[DEFAULT]: Data stream stopped.//
//Cyton: startDataTransfer(): writing 's' to the serial port...//
//Hub: sendCommand(char): sending 's'//
//Hub: processCommand: success -- 130320/

Thie contect of the files is as following:

/%OpenBCI Raw EEG Data//
//%Number of channels = 16//
//%Sample Rate = 125.0 Hz//
//%First Column = SampleIndex//
//%Last Column = Timestamp //
//%Other Columns = EEG data in microvolts followed by Accel Data (in G) 
interleaved with Aux Data//
//0, 2563.48, 1098.63, 1098.63, 1098.63, 1098.63, 1098.63, 1185.89, 
-177612.31, -93378.03, -187500.02, -187500.02, -187500.02, -187500.02, 
-187500.02, -187500.02, -187500.02, -4.096, 0.000, 0.000, 23:30:03.984, 
1569191403984/

//
cat 
/home/djan/Documents/OpenBCI_GUI/Recordings/OpenBCISession_2019-09-22_23-28-38/OpenBCI-RAW-2019-09-22_23-29-21.txt
/%OpenBCI Raw EEG Data//
//%Number of channels = 16//
//%Sample Rate = 125.0 Hz//
//%First Column = SampleIndex//
//%Last Column = Timestamp //
//%Other Columns = EEG data in microvolts followed by Accel Data (in G) 
interleaved with Aux Data//
//0, 2563.48, 1098.63, 1098.63, 1098.63, 1098.63, 1098.63, 1164.26, 
184204.11, -93378.03, -187500.02, -187500.02, -187500.02, -187500.02, 
-187500.02, -187500.02, -187500.02, -4.096, 0.000, 0.000, 23:29:22.090, 
1569191362090/

retiutut, 2019-09-22:
@djan4848 Try using GUI 4.1.2 now just to see if it works.

djan4848, 2019-09-22:
Same behavior:

/djan@Base:~/DATA/Investigaciones/OpenBCI_GUI_4.1.2$ ./OpenBCI_GUI//
//Screen Resolution: 1920 X 1080//
//Welcome to the Processing-based OpenBCI GUI!//
//For more information, please visit: 
https://docs.openbci.com/OpenBCI%20Software///
//Launching application from local data dir//
//OpenBCI_GUI: hubStart: OS Detected: Linux//
//Local Version: 412, Latest Version: 415//
//GUI needs to be updated. Download at 
https://github.com/OpenBCI/OpenBCI_GUI/releases/latest//
//OpenBCI_GUI::Setup: Is RX mulitcast: false//
//OpenBCI_GUI::Setup: Has RX joined multicast: false//
//ControlP5 2.2.6 infos, comments, questions at 
http://www.sojamo.de/libraries/controlP5//
//The file "SavedData/Settings/UserPlaybackHistory.json" is missing or 
inaccessible, make sure the URL is valid or that the file has been added 
to your sketch and is readable.//
//OpenBCI_GUI::Control Panel: Playback history file not found or other 
error.//
//Instantiating hub object...//
//Hub: clientEvent: handshake complete//
//Refreshing port list...//
//sourceList: click! //
//setInterface: curInterface: -1//
//channel count set to 8//
//Setting Latency to 10000//
//Setting WiFi Internet Protocol to tcp//
//Setting WiFi style to dynamic//
//[DEFAULT]: Protocol Serial Selected for Cyton//
//setInterface: curInterface: 0//
//Setting sample rate for Cyton to 250Hz//
//
//
//sample rate set to: 250//
//[DEFAULT]: Transfer Protocol set to serial//
//serialList: click! //
//[DEFAULT]: OpenBCI Port Name = /dev/ttyUSB0//
//channel count set to 16//
//Calling initSystem()//
//
//
//=================================================//
//||             INITIALIZING SYSTEM             ||//
//=================================================//
//
//cyton: openNewLogFile: opened ODF output file: 
SavedData/OpenBCI-RAW-2019-09-22_23-53-43.txt//
//"Arial Bold" is not available, so another font will be used. Use 
PFont.list() to show available fonts.//
//init analog read bar 0//
//init analog read bar 1//
//init analog read bar 2//
//Hub: processConnect: made it -- 33970 code: 200//
//Hub: setBoardType(): sending 'daisy -- 33970//
//Hub: processBoardType: success -- 33970//
//[SUCCESS]: The GUI is done initializing. Settings Loaded! Press "Start 
Data Stream" to start streaming! --  Cyton firmware v3.1.2//
//InitSettings: Saving Default Settings to file!//
//SoftwareSettings: 3 active widgets saved!//
//The file "SavedData/Settings/DaisyUserSettings.json" is missing or 
inaccessible, make sure the URL is valid or that the file has been added 
to your sketch and is readable.//
//InitSettings: SavedData/Settings/DaisyUserSettings.json not found or 
other error.//
//[DEFAULT]: The new data source is LIVE (from Cyton  v3.1.2) and NCHAN 
= [16]. User settings error: Default Settings Loaded.//
//[DEFAULT]: Data stream started.//
//Cyton: startDataTransfer(): writing 'b' to the serial port...//
//Hub: sendCommand(char): sending 'b'//
//Hub: processCommand: success -- 41400//
//Hub: apparent sampleIndex jump from Serial data: 0 to  56. Keeping 
packet. (1)//
//numPacketsDropped = 56//
//[DEFAULT]: Data stream stopped.//
//Cyton: startDataTransfer(): writing 's' to the serial port...//
//Hub: sendCommand(char): sending 's'//
//Hub: processCommand: success -- 51564//
//[DEFAULT]: Data stream started.//
//Cyton: startDataTransfer(): writing 'b' to the serial port...//
//Hub: sendCommand(char): sending 'b'//
//Hub: processCommand: success -- 53492//
//[DEFAULT]: Data stream stopped.//
//Cyton: startDataTransfer(): writing 's' to the serial port...//
//Hub: sendCommand(char): sending 's'//
//Hub: processCommand: success -- 62656//
//[DEFAULT]: Data stream started.//
//Cyton: startDataTransfer(): writing 'b' to the serial port...//
//Hub: sendCommand(char): sending 'b'//
//Hub: processCommand: success -- 64352//
/

retiutut, 2019-09-26:
@BCIStorm Combining your issue with this one!

retiutut, 2019-09-28:
@djan4848 Try this  from #608 :
> Richard and BCIstorm, YES, a V1 Cyton has the old RFduino firmware. All three processors must be updated in the EXACT sequence specified.

https://github.com/OpenBCI/OpenBCI_Cyton_Library/blob/master/UPGRADE_GUIDE.md

djan4848, 2019-09-29:
RFDuino is gone in Arduino ide 1.8.10 there is no such board on "board 
manager", I don't know how to deal with it. :-(. Should I have to use an 
older ide version to update the RadioDevice32bit ?? In such case, which 
is the latest ide version to use tu update RFDuino?

Regards

wjcroft, 2019-09-29:
Use IDE 1.5.8 beta as described in the doc:
https://docs.openbci.com/docs/02Cyton/CytonRadios

retiutut, 2019-10-13:
@djan4848 Did you try to re-upload the firmware to the Cyton board? What's the status of this issue?

retiutut, 2019-10-13:
@djan4848 Did you try to re-upload the firmware to Cyton and update the Dongle? What's the status of this issue?

ra75763, 2019-10-13:
The problem is solved and I replied the steps which I took to solve.

djan4848, 2019-10-14:
Sorry for the delay to reply, I'm outside in a travel. I'll retry to 
update the Dongle on 18th october when I'll be back.

djan4848, 2019-10-18:
Hi Richard

I did the update to the Dongle and this is the new behavior:

  * Cyton without Daisy works fine with OpenBCI_GUI 4.1.5 that means 8
    channels.
  * Cyton with Daisy (16 channels) get red light blocked on the Dongle
    and GUI freezed

As I have acquired another Cyton board his year, I moved my Daisy to the 
newer Cyton and  both works fine.

That means, with new board + old Daisy all is working fine with GUI 
4.1.5. I keep the old Cyton for experiments were 8 channels are enough.

Even if it is not a totally succesful story, it is ok for me as I keep 
both cyton working with new GUI.

Thanks for your help.

Regards

Damian

retiutut, 2019-10-18:
@djan4848 I will go ahead and close this issue now. Glad to hear you got it working!

closed 2019-10-18 by retiutut

davidmarcelbaum, 2022-11-03:
> […](#)

Hello Damian,

I am in a similar situation. How did you manage to get the register information from the serial port, please? I am more a neuroscientist than a computer engineer, but I am able to follow instructions.

I would appreciate any help!
Regards
David