OpenBCI/OpenBCI_GUI #603: No Signal from electrodes and accelerometer even though Ganglion is connected

lsteinb, 2019-09-22:
## Problem

I've successfully connected the ganglion board to gui via bluetooth, but when I try to start data acquisition, signals stays zero. I'm not even seeing noise or anything. In addition after turning on the accelerometer, it also doesn't show any change when movin the board. It stays zero all the rime. 

## Expected 

See signal in gui after starting data stream

## Operating System and Version

Windows

## GUI Version
v4.1.5

## Type of OpenBCI Board

Ganglion

## Are you using a WiFi Shield?

No

retiutut, 2019-09-22:
Please provide pictures and a console log.

retiutut, 2019-09-23:
If we suspect the hardware is broken, we can request a replacement from support@openbci.com.

lsteinb, 2019-09-23:
[Console_2019-09-22_17-08-38.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/3640854/Console_2019-09-22_17-08-38.txt)
[image]

lsteinb, 2019-09-23:
I think it's really HW part issue, since it performs the impedance check correctly. And also I noticed that when I touch he board a bit on some places it transmits some signals, but no accelerometer, and after a while it gets stuck.

lsteinb, 2019-09-23:
[Console_2019-09-22_17-08-38.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/3640870/Console_2019-09-22_17-08-38.txt)

retiutut, 2019-09-23:
@lsteinb Thanks so much for the full console log! Looks like hardware issues. This bit right here happens two times in a row:

```
[DEFAULT]: Data stream started.
Ganglion: startDataTransfer(): sending 'b
Hub: sendCommand(char): sending 'b'
Hub: processCommand: success -- 342885
OpenBCI_GUI: openNewLogFile: opened ODF output file: C:\Users\r&d\Documents\OpenBCI_GUI\Recordings\OpenBCISession_2019-09-22_17-13-53\OpenBCI-RAW-2019-09-22_17-14-19.txt
Hub: apparent sampleIndex jump from Serial data: 0 to  3.  Keeping packet. (183)
numPacketsDropped = 2
Hub: apparent sampleIndex jump from Serial data: 10 to  15.  Keeping packet. (184)
numPacketsDropped = 4
Hub: apparent sampleIndex jump from Serial data: 6 to  11.  Keeping packet. (185)

ETC...
ETC...
```

closed 2019-09-23 by retiutut