OpenBCI/OpenBCI_GUI #606: EEG raw data ploting from csv file

shama992, 2019-09-24:
please guide me how u can plot raw data EEG signals graph in your project.

retiutut, 2019-09-24:
This is done in TimeSeries.pde using GPlot from the Graphica library: https://jagracar.com/grafica.php

Otherwise, look at the data in the CSV and parse the data line by line. First column is the sample count.

For CSV files that have been converted from an SD card recording, there is no timestamp at the end:
```
7,-63526.492,-48039.53,-73942.72,-62922.414,-121932.0,-73300.82,-154977.95,-165474.86,-49437.766,-62449.9,-89375.234,-76500.5,-143431.27,-128037.36,-146416.58,-17027.246
```
- 7 = sample number
- Followed by 16 channels of raw data in microvolts

There will also be some lines of data that have accelerometer readings:
```
8,-63730.832,-48231.04,-73833.76,-62920.72,-121889.11,-73333.12,-154973.28,-165498.39,-49470.758,-62552.984,-89367.88,-76533.71,-143428.44,-128123.75,-146341.11,-16972.506,0.004,0.91,0.404
```
- 8 = sample number
- Followed by 16 channels of raw data in microvolts
- Ends with 3 channels of accelerometer data

closed 2019-09-24 by retiutut

shama992, 2019-09-25:
thanks retiutut but i dont have knowledge about time series can you please explain about that how its work